OrcaSlicer/OrcaSlicer #7936: Fuzzy skin painting tool, like in PrusaSlicer

VardanK, 2025-01-06:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Fuzzy skin often ends up being applied in unintended areas, such as threads or moving parts, which can cause functional issues. While modifiers can help in controlling this to some extent, they are not always convenient—especially when you only want to add fuzziness to a single face.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

PrusaSlicer 2.9 introduced an innovative feature for adding fuzzy skin by simply painting directly on an object. This approach is already impressive, but the real highlight is the "Smart Fill" functionality. With it, users can apply fuzzy skin exclusively to specific faces, enabling targeted customization. This capability allows designers to create dedicated faces during the design process specifically for fuzzy skin application.

I suggest implementing a similar feature in OrcaSlicer. Adding the ability to paint and apply fuzzy skin to specific faces, including a "Smart Fill" option, would significantly enhance customization and user experience.

### Additional context

Screenshot done from - [Multiple Beds, Scarf Seams & Fuzzy Skin Painting! (PrusaSlicer 2.9)](https://www.youtube.com/watch?v=XLcoLCAhRks)

[image]

jonathanstarke, 2025-01-30:
Would love this as well!

tlhintoq, 2025-02-02:
Fuzzy skin texture painting tool is clearly a desired item.  Just sayin'

https://github.com/SoftFever/OrcaSlicer/issues/7936
https://github.com/SoftFever/OrcaSlicer/issues/4471
https://github.com/SoftFever/OrcaSlicer/issues/3330

WainingForests, 2025-02-25:
Please <3

thewolfman56, 2025-02-25:
Another vote for this, please

QuietJJ, 2025-02-25:
Another vote for Paint on fuzzy skin! It would make a massive difference to things.

Dustmuffins, 2025-02-27:
Would love this feature! It would be so useful for a lot of my projects!

kdm303, 2025-03-01:
This feature would be most welcome.

Funnysisters, 2025-03-14:
И я за это улучшение!

KoukiMyke88, 2025-03-26:
please please please add this feature

VardanK, 2025-03-26:
Dear @SoftFever, Thank you for the Gigantic effort you put into OrcaSlicer. I am writing to bring your attention to this feature request. I suspect you are overburdened with development tasks, but is there any chance you will include this feature in upcoming releases?

Alexx40in, 2025-03-26:
Vote for this, please

gusarg81, 2025-03-26:
I would love to have this as Prusa Slicer does. I use Orca for all my printers and I Design and Print Prescription Frame Glasses. So I use fuzzy for all the frame but I wan to avoid some critical parts.

SoftFever, 2025-03-29:
Yes, this is planned.

gusarg81, 2025-05-20:
> Yes, this is planned.

Any upcoming version of Orca Slicer or there is not yet a timeline?

closed 2025-07-18 by Noisyfox