OrcaSlicer/OrcaSlicer #7937: Problem on linux with .AppImage

onlyymax, 2025-01-06:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

CPU: Intel Core I9-12900K  
GPU: NVIDIA RTX 3070 (custom ROG)  
RAM: DDR5 32GB  

I always start Orca by running a script that first executes these commands:

export __EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json  
export WEBKIT_DISABLE_DMABUF_RENDERER=1  
export WEBKIT_FORCE_COMPOSITING_MODE=1  
export WEBKIT_DISABLE_COMPOSITING_MODE=1  

I recently noticed that with the first command, Orca starts lagging and slowing down. If I remove it, it works correctly. It's always a hassle to run these commands beforehand. I can't wait for you to upload Orca Slicer to Flathub, so it can resolve all our problems as Linux users.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Flathub

Cybereu, 2025-01-16:
Unfortunately I'm having the same problem with the flatpak nightly builds, atual remedy is using mesa drivers:
env __EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json

onlyymax, 2025-01-16:
yes but it lags like hell when I try to slice something heavy, sometimes it doesn't even work and fedora crashes

onlyymax, 2025-01-16:
I hope they fix everything and upload it to flathub

jarrard, 2025-01-27:
Using 50_mesa.json just blanks the home window out for me and renders no model.  At least for flatpak versions.

coryrc, 2025-12-25:
@onlyymax Flatpak exists now, so can you close this Issue?

onlyymax, 2025-12-25:
sure

closed 2025-12-25 by onlyymax