OrcaSlicer/OrcaSlicer #7941: OrcaSlicer 2.3.0-dev - Camera washed out/ overexposed

wwian, 2025-01-06:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.3.0-dev

### Operating System (OS)

Windows

### OS Version

Windows 11 Home 23H2

### Additional system information

Processor	Intel(R) Core(TM) i7-8550U CPU @ 1.80GHz   1.99 GHz
Installed RAM	16.0 GB (15.9 GB usable)
System type	64-bit operating system, x64-based processor

### Printer

Bambu Lab X1 Carbon

### How to reproduce

1. open orcaslicer
2. click 'device'
3. under 'camera' click play button
[image]
[image]
[image]

### Actual results

The camera image is messed up, washed out, and overexposed.

### Expected results

The camera image is clear and viewable. 

### Project file & Debug log uploads

[ORCA-Log_conf.zip](https://github.com/user-attachments/files/18322062/ORCA-Log_conf.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Anything else?

This issue happens reguardless of what project file is open.

sschwetz, 2025-01-08:
This is also happening on the MacOS release as well

wwian, 2025-01-20:
This is a video taken of the camera issue.
https://github.com/user-attachments/assets/1f2a9a78-7d6c-40f0-ae3b-4a0aa87e7a50

closed 2025-04-29 by github-actions[bot]