OrcaSlicer/OrcaSlicer #7942: 2.3.0 DEV After 6 prints, OrcaSlicer fails to send print job

buckeyestargazer, 2025-01-06:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.3.0 DEV

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Bambu P1S, Bambu X1C

### How to reproduce

Send 6 individual print jobs to the printer.  The 7th one will fail to send with no explanation.

### Actual results

The 7th print job fails.

### Expected results

The print should send successfully to the printer.

### Project file & Debug log uploads

[debug_Wed_Jan_01_11_13_33_236404.log.zip](https://github.com/user-attachments/files/18322503/debug_Wed_Jan_01_11_13_33_236404.log.zip)
[Cable-C11.RASA11-ALDew-78mm.zip](https://github.com/user-attachments/files/18322515/Cable-C11.RASA11-ALDew-78mm.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

buckeyestargazer, 2025-02-08:
Is anyone else seeing this?  Using the nightly builds, after 6 prints (either sent to multimachine or individually), I can't print with an error simply "failed to print."  I have to restart OrcaSlicer to print again.

buckeyestargazer, 2025-05-10:
This issue is not stale for me!  It doesn't matter what version of OrcaSlicer I use.  I can only send exactly 6 prints to my printers before I get an error "failed to print" with no other information.  I restart orcaslicer, and six prints later it does the same thing.  It's been months like this...

buckeyestargazer, 2025-05-22:
This issue is getting really frustrating.  I'm about ready to move on from OrcaSlicer which I don't want to do.  I understand this may just be an issue on my machine, but I have tried everything I can think of.  It's been nearly 5 months since I first reported this and it would be nice to at least receive an acknowledgment of some kind.  This is the only error I receive, with no info as to what is happening.

[image]

Schluggi, 2025-08-17:
Same issue with the current alpha of 2.3.1