OrcaSlicer/OrcaSlicer #7944: Disbale/reduce Fan speed for certain Layer(s).

AnEn030, 2025-01-06:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

When printing with ABS/ASA the cooling is helping to increase the quality of the print BUT the too high temperature difference between previous layers may decrease layer adhesion. Especially if the print area is strongly decreaseing from the previous one. I had the issue with ASA and the Voron 0.1 tophat. I fixed it by manually editign the gcode. 
Before over layer was "cooled" and the "connecting" Top surface had very bad layer adhesion. 
[image]

### Which printers will be beneficial to this feature?

Klipper

### Describe the solution you'd like

option A.: hight range modifer where cooling is included can be set ( eg. % Value)

B.: usage of "Paint on" "fan" reduction.... .

### Describe alternatives you've considered

1. manually editing. ...
2. maby with "add modifiers"

### Additional context

[image]

buzzhuzz, 2025-01-07:
Shouldn't cooling settings in filament profile address this concerns?
You may set cooling to 0% for layers which takes longer than some threshold.

[image]

AnEn030, 2025-01-07:
not really cause i want to have only  the last 1 or 2 top layers to be no/reduced fanspeed of the larger area and than the one or two upcoming layer of the reduced area to be with no fan.

Aside from the additional image, i added my manually edited g code as zip file.
[image]
[Lower_Corner_1_and_3_Clip_x2_21m_0.25mm_240C_ASA_Voron_v0_120.zip](https://github.com/user-attachments/files/18337962/Lower_Corner_1_and_3_Clip_x2_21m_0.25mm_240C_ASA_Voron_v0_120.zip)