OrcaSlicer/OrcaSlicer #7947: Ignoring Motion Ability Maximum speed values

Awesten3D, 2025-01-07:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2

### Operating System (OS)

Windows

### OS Version

Windows 10

### Additional system information

CPU: i9 1200k
RAM: 32gb
GPU: RTX 3090

### Printer

CR-6 SE

### How to reproduce

Set a maximum speed under motion abilities to 10mm/s for the Z axis. Set Travel at 200mm/s. Slice the file. Slicer ignores the maximum speeds set in the machine motion abilities section and uses travel speed instead, during z hop moves this causes missed steps and the nozzle to get progressively lower and lower into the build plate/ printed part. Follows the height if I set a minimum z hop start height in machine settings.
[image]

[image]

### Actual results

Printer uses travel speed for z hops and causes skipped steps. Have spent 4 days ruling out mechanical problems and can reproduce the same issues on my Ender 3 which is a very similar style printer.

### Expected results

Slicer should limit max z speed to whats set in slicer to avoid skipping steps and accuracy issues.

### Project file & Debug log uploads

[CRFH_pin_v2.zip](https://github.com/user-attachments/files/18326660/CRFH_pin_v2.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Anything else?

MxBrnr in the orca slicer discord helped me troubleshoot and come to a conclusion. He reproduced the same slicer issue on his end.

Matt-Pengelly, 2025-02-11:
@Awesten3D I believe I am getting the same issue and am using v2.2.

I am not certain its actually an issue with OrcaSlicer though, because the firmware (Klipper in my case) should be limiting the z acceleration and speed to a safe value. Maybe these limits get overwritten by Orca but I dont see where this is happening in the gcode.

Have you 100% confirmed the z axis is not skipping steps?

Awesten3D, 2025-02-11:
@Matt-Pengelly Yes I have confirmed that they ARE skipping steps, which is where the problem arises. They do however run in spread cycle but your issue may be somewhere else as if you have Klipper enabled in orca it should ignore the limits/not override your current max velocity in your firmware. I have since switched to klipper on this specific printer and it cured my issue however I feel this is still a problem as there should be a z-hop speed setting like in most other slicers.

Matt-Pengelly, 2025-02-11:
Sorry I meant to say skipping steps under normal conditions e.g. when moving the z axis manually in the interface, or when printing with a different slicer.

That's interesting that switching to klipper fixed it. To me that sounds more like a firmware configuration issue relating to z axis limits than a problem with the generated gcode.

Awesten3D, 2025-02-11:
So am I misunderstanding the "emit to gcode" section where it says its ignored if klipper is the selected flavour of gcode? Ignored as in it will or will not use the limits set in the printer profile?

Sineos, 2025-02-11:
Klipper's behavior is as follows:
* `max_velocity` and `max_accel` from the `printer.cfg` are overridden by the `SET_VELOCITY_LIMIT` command and Orca is using this
* `max_z_velocity` and `max_z_accel` default to `max_velocity` and `max_accel` if not set. If set, they are not overridden.

Awesten3D, 2025-02-12:
This can be closed now! My issue has been solved by switching to klipper and using firmware to limit max_z_velocity and max_z_accel. I am still convinced that there should be a separate speed for z hop but there's plenty of work arounds! Thanks for the help.

closed 2025-02-12 by Awesten3D

Laihela, 2026-11-13:
Had [this](https://www.reddit.com/r/3Dprinting/comments/1v6ttmf/comment/19slh7w8th5yb/?context=1&persist-lb-on-conversion=true&screen_view_count=2) happen to me today, so this is still an issue. The z axis tries to move so fast that the motor stalls and the next object is 'printed' in the air. I don't think expecting the user to change the printer firmware is a valid solution, the slicer should obey it's own speed limit settings.