OrcaSlicer/OrcaSlicer #7953: Problem adding custom filament

GlauTechCo, 2025-01-07:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0, 2.3.0 dev

### Operating System (OS)

macOS

### OS Version

14.2.1 Sonoma

### Additional system information

M4 mini

### Printer

FlashForge M5p

### How to reproduce

I added the problem as a video

### Actual results

I added the problem as a video

### Expected results

I added the problem as a video

### Project file & Debug log uploads

https://github.com/user-attachments/assets/4fe3ab02-ccad-4204-8cb3-d5cbc4374856

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Hammerfest, 2025-01-07:
That explains so much, TY for the fix.

locki-cz, 2025-01-07:
> orca doesnt recognize capitalized "Nozzle" in your printer name. I'll submit a fix later

it probably not recognize more than "" :) what is wrong with my profile name?

[image]

GlauTechCo, 2025-01-07:
@Azio-Pantheon Thnx.
[image]

Hammerfest, 2025-01-07:
> it probably not recognize more than "" :) what is wrong with my profile name? [...]

per the commit:
"Create a lowercase version of the printer_name for case-insensitive search"
So looks like an issue where the search should not be case-insensitive, but because it is, making the printer name lower case resolves it as a workaround.

(IANAD)

Hammerfest, 2025-01-07:
> I prefer having clear documentation on the proper naming convention, so we don't have to deal with capitalization/space/bracket in the printer name. We can't expect Orca to catch all the cases until we have a standard on naming printer profiles.

Your not wrong, I myself am not a fan of long overly complicated printer names for base profiles either, BUT my comment was more about the fact a simple case trap has caused this bug.

An example is my A1, but this extends to all printers I have
BBL-A1_H0.4mm-Artemis
BBL-A1_B0.4mm-Artemis
BBL-A1_H0.4mm-Artemis-test

Hammerfest, 2025-01-07:
> > An example is my A1, but this extends to all printers I have BBL-A1_H0.4mm-Artemis BBL-A1_B0.4mm-Artemis BBL-A1_H0.4mm-Artemis-test
> 
> Sorry, I feel like I am still missing your point. Can you explain more on what's up with your examples?
> 
> From what I can tell, none of your examples will show up for custom filament because they don't have the word "nozzle" in them

The example was just a reference both for the fact it wasn't showing up for me due to my naming, as well as how the clear documentation on a proper naming convention would only apply to defaults not something like myself or others use.

PR #7965 (2025-01-08): fixed a custom filament creation bug
Orca doesnt recognize capitalized letter in 'Nozzle' in the printer name. Those printer profiles will not show up when creating custom filaments, which blocks users the ability to create custom filament profiles for those printers.

This fixes #7953 

# Screenshots/Recordings/Graphs
2.3.0-dev:
[image]

after:
[image]

## Tests

Tested locally on Windows11
closed 2025-01-08 by Noisyfox