OrcaSlicer/OrcaSlicer #7961: Adding a 3d Printer profile - Anycubic Kobra S1

ManethonSega, 2025-01-07:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

It is a new printer and its profile is not available on Orca slicer

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

the Printer profile is already available on Anycubic slicer next, which is made according to orca slicer. this profile could be exported.

bistory, 2025-01-27:
Hi, it has been added in #7987

Schrolli91, 2025-01-27:
Clearing the `Appdata` folder did the trick - thanks

brosskgm, 2025-02-27:
It was in the 2.3.0 Beta Appimage on Linux machine. Ran good.

Schrolli91, 2025-03-12:
So I think, this can be closed, because it will be included in the next release.

brosskgm, 2025-03-15:
I haven't received my S1 Combo yet. Maybe another week. But Anycubic web site does show USB, WiFi and Network connection available as well as an app for the phone. Most others Cura etc. It been said does not work with it. They don't have the printer in their setup. I looked. I also checked ideaMaker that I use with my other printers and they don't have it either. The latest release of OrcaSlicer 2.3 rc is available and is the only so far that has S1 Combo in it.

michas79de, 2025-03-15:
Reading "Ran good.", I thought you mean with the S1 already. Thanks though!
I've found no clear confirmation. So I'll have to hope the direct wifi connection between OrcaSlicer and the S1 will work until I get mine. And the ACE Pro also...

brosskgm, 2025-03-15:
The software 2.3 beta ran good on Ubuntu. The 2.3 rc also. The previous versions were hit and miss. With testing the slicing with S1 Combo(Ace Pro) multi colors it did well.

Schrolli91, 2025-03-15:
> But Anycubic web site does show USB, WiFi and Network connection available as well as an app for the phone.

Direct printing over Network from Orca is not possible (only with Anycubics own Slicer) - just with a little Workaround.
My S1 is here and I tried Rinkhals. That works great for now and i can directly print trough Orca.
https://github.com/jbatonnet/Rinkhals

michas79de, 2025-03-25:
> Direct printing over Network from Orca is not possible (only with Anycubics own Slicer) - just with a little Workaround. My S1 is here and I tried Rinkhals. That works great for now and i can [...]

How do I configure Rinkhals in Orca Slicer?

Now I have a Kobra S1 with the latest firmware and the latest rinkhals.
The printer is in LAN mode.
I have the latest Orca Slicer 2.3.0 in Ubuntu.
I entered the IP of the printer and clicked TEST which was successful, but then (after OK) OrcaSlicer crashed and kept crashing after restarting until I deleted the OrcaSlicer .config folder.
[image]
[image]

In the browser I can see several UIs on the printer's IP on ports 80 and 4409 (both Mainsail) and 4408 (fluidd).
I have tried all of them in Orca without success, it's still crashing.

IDK what's the problem. Any hints?

smougenot, 2025-04-30:
You also have the Home Assistant integration 
https://github.com/WaresWichall/hass-anycubic_cloud

Schrolli91, 2025-06-09:
You have to create four rolls of filament in Orca (regardless of whether they are physically loaded in the ACE). The number is mapped to the slot in the ACE Pro.

xorl, 2026-01-18:
> You have to create four rolls of filament in Orca (regardless of whether they are physically loaded in the ACE). The number is mapped to the slot in the ACE Pro.

How do you create a roll like this to map the roll? I am failing at google right now..

michas79de, 2026-01-19:
In the meantime there are multiple nozzle diameters in Anycubic Slicer Next:

[image]

So an update would be great!