OrcaSlicer/OrcaSlicer #7970: Reordering objects does not work

blonestar, 2025-01-08:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Voron Trident / v2.4

### How to reproduce

1. Create or add at least 3 object on the plate.
2. Switch to Object 
3. Reorder object so you have the one first to print on the top

### Actual results

Slicer ignores order of objects and print objects on **the first layer** in **its own order**.

### Expected results

Object on top should be printed first on first layer, and to respect the object order.

### Project file & Debug log uploads

[Orca object ordering problem example.zip](https://github.com/user-attachments/files/18340341/Orca.object.ordering.problem.example.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Anything else?

[image]

blonestar, 2025-01-08:
I had to open a new bug report for the same issue, as it still hasn't been fixed in the latest version of the Orca slicer.
https://github.com/SoftFever/OrcaSlicer/issues/2270

Noisyfox, 2025-01-08:
[image]

blonestar, 2025-01-08:
omg, how come I don't know about that??
Thank you for enlightening me :D

closed 2025-01-08 by blonestar