OrcaSlicer/OrcaSlicer #7975: Device > Camera not working when MacOS Internal Firewall is active

SC-JBG, 2025-01-08:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.3.0-dev

### Operating System (OS)

macOS

### OS Version

15.2 (24C101)

### Additional system information

MacBook Pro 16" 2023
Apple M2 Max
32 GB RAM

### Printer

Bambu Lab X1-Carbon

### How to reproduce

1. Have the Internal Firewall of MacOS turned ON (Settings --> Network --> Firewall --> Toggle ON)
2. Open OrcaSlicer
3. Go to "Device" tab
4. Start "Camera" using the Play-Button

### Actual results

With turned on internal firewall under MacOS 15.2 the integrated OrcaSlicer player, that shows the camera feed, stops from playing state to stopped state and shows an error message:

`Player is malfunctioning. Please reinstall the system player. |2:2]`

If you do not do anything and wait for a little while it will switch back from the error message to "Loading..." but without ever showing the actual camera feed.

### Expected results

The player should play the printers video feed from the printer webcam.

### Project file & Debug log uploads

Project file: [ProjectFile.zip](https://github.com/user-attachments/files/18347452/ProjectFile.zip)

Logs: [Logs.zip](https://github.com/user-attachments/files/18347454/Logs.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

- The issue only happens when the MacOS internal firewall is enabled (Settings --> Network --> Firewall --> Toggle ON).
- If you turn off the firewall entirely, the video feed is loading as expected.
- Creating an exception rule for OrcaSlider and BambuStudio under the Firewall options does not fix the problem. The firewall must be turned off completely.
- If you open OrcaSlicer and start the video feed while the firewall is turned on, you'll get the error message above. Once you turn the firewall off the video feed will automatically start again, without restarting or anything.+
- This issue is reproducable in the latest stable version of OrcaSlider, BambuStudio and the Dev-Build of OrcaSlicer.
- It began after upgrading the OS version to MacOS 15.2 and worked fine before in older versions.
- The video feed is working in e.g. the BambuHandy app as expected.

buzzhuzz, 2025-01-08:
*I'm actually does not have camera-enabled setup and never looked into video playback implementation in OrcaSlicer, just sharing my thoughts based on my background knowledge*

The problem lies in the RTSP protocol and how it works:
# PC connects to the camera to TCP port 554
# PC communicates with the camera setting up configuration and starting video stream
# Camera then starts sending stream of UDP packets to a randomly selected UDP port in a range about 6900-7200 (may vary depending on streamer implementation)

So, with firewall enabled, your OS rejecting incoming UDP packets with media data making player raising error.

In order to workaround this, the client shall request media transfer over TCP in their SETUP request. In this case, the same TCP connection will be reused for control and media data. However, one should know that TCP is an optional per RTSP standard and because of this not every camera/server support streaming over TCP.

I see solution to this issue as an additional configuration item which would let user choose between TCP and UDP transport for RTSP.

buzzhuzz, 2025-01-08:
I've took a quick look into the sources and see that Bambu printers are special since video stream is coming through their binary libraries hiding video streaming protocol implementation. Because of this it is hard to say if my comment above applicable to BambuLab printers (are they using UDP streaming or not) and if it is possible to configure video streaming protocol for them.

I'd suggest you to reproduce the same issue using BambuStudio and make a clone of this issue in their tracker.

SC-JBG, 2025-01-08:
> I'd suggest you to reproduce the same issue using BambuStudio and make a clone of this issue in their tracker.

Good idea. I checked their issues log and there are already several issues reported. I added a comment to the main ticket here: https://github.com/bambulab/BambuStudio/issues/5642

closed 2025-04-16 by github-actions[bot]