OrcaSlicer/OrcaSlicer #7980: Part Fan speed is not Interpolating between Min/Max Fan setting for Layer Times

jeffalanmcclain, 2025-01-08:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Voron Trident

### How to reproduce

1. In Filament Settings -> Cooling
2. Under Part cooling fan
3. For Min fan speed threshold->Set Fan speed to 30% and Layer time 90s
4. For Max fan speed threshold->Set Fan speed to 100% and Layer time 10s
5. Check Slow printing down for better layer cooling
6. Set Min print speed -> 15mm/s
7. Slice a 3DBenchy

### Actual results

Depending on your line speed settings, you will have different layer times, however, you should be able to copy multiple benchy's or speed/slow settings down to get some layer times that will fall between the 10s and 90s.  For mine, even when I slice and view the layer times and have plenty of things between 30s and 50s, the part fan shows 99% and is NOT interpolating between the min/max settings like it says it should.

### Expected results

I would expect when there are individual layer times between the min/max fan threshold layer times, that you would have proper interpolation between the fan % settings.  But in these cases it is ALWAYS solid at 99% (this is NOT related to bridging or overhang, as those get a proper 100%).

### Project file & Debug log uploads

[debug_Wed_Jan_08_15_21_02_207784.log (2).zip](https://github.com/user-attachments/files/18353871/debug_Wed_Jan_08_15_21_02_207784.log.2.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

[image]
[image]
[image]

MxBrnr, 2025-01-08:
I believe this issue was fixed by the time OrcaSlicer v2.2.0 was released. Consider updating to the latest version and then re-testing for the fault. Fixes are not implemented on outdated versions.

Openwrtfunboy, 2025-01-17:
+1 pls fix

ksummers92, 2025-02-13:
I can confirm that this is still a problem in 2.2.0. See post here: https://discord.com/channels/1137181739773603922/1339590800291663969/1339590800291663969

closed 2025-05-22 by github-actions[bot]

pm0u, 2025-06-27:
Seems to be happening to me on 2.2.0

[image]

[image]

[image]

layer time is 140s, over 3x the max layer time and is still at 20% when in reality it should be at the min, 5%.

pm0u, 2025-06-27:
Still present on 2.3.1-dev

PhilBaz, 2025-08-31:
Thats is mine...Ill close it if that's better. id like to add there also seems to be a problem with 

Overhangs and external bridges fan speed

Internal bridges fan speed

Support interface fan speed

They simply don't seem to respect the settings.

ehtrinkl, 2025-12-26:
Hi, 

I created a new issue confirming this issue still exists in latest release.