OrcaSlicer/OrcaSlicer #7982: Application crash in Multimaterial Tab

GlauTechCo, 2025-01-08:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0, 2.3.0 dev

### Operating System (OS)

macOS

### OS Version

14.2.1 Sonoma

### Additional system information

M4 Mac Mini

### Printer

FlashForge M5p

### How to reproduce

1. Open Printer settings
2. Open Multimaterial tab
3. Extruders number 0

### Actual results

When I reduce the number of extruders to 0, the application crashes.

https://github.com/user-attachments/assets/ca598e3e-5e5c-46d1-8ad0-2d1ea8234657

### Expected results

Reducing the number of extruders below 1 is prevented, or its collapse should be prevented with a warning.

### Project file & Debug log uploads

Add video

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Noisyfox, 2025-01-09:
Seems to be a macos specific bug... On windows you cannot set the extruder number to 0.

Noisyfox, 2025-01-11:
https://github.com/SoftFever/OrcaSlicer/blob/e3155861258205f93048bb9c10c3fae511bd6b15/src/slic3r/GUI/Field.cpp#L1067-L1076

The issue came from here. The min value constraint doesn't work on macos.

PR #8009 (2025-01-22): Fix crash on macOS that you could set extruder number to 0
Fix #7982
cherry picked from commit prusa3d/PrusaSlicer@555193f6ec477d4107f8998c7a3c441cde72382c

Thanks Prusa!
closed 2025-01-12 by GlauTechCo