OrcaSlicer/OrcaSlicer #7983: Single extruder multi filament prime tower problems

GlauTechCo, 2025-01-08:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0, 2.3.0 dev

### Operating System (OS)

macOS

### OS Version

14.2.1 Sonoma

### Additional system information

M4 Mac Mini

### Printer

FlashForge M5p

### How to reproduce

Since there were so many mistakes, I explained them in the video.

### Actual results

Since there were so many mistakes, I explained them in the video.

### Expected results

https://github.com/user-attachments/assets/c6f0c550-ef33-4409-b72f-570719dc8050

### Project file & Debug log uploads

Ok

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Noisyfox, 2025-01-11:
Could you upload a 3mf file that could reproduce this problem?

GlauTechCo, 2025-01-12:
> Could you upload a 3mf file that could reproduce this problem?

added

[Orca Stanford Bunny.3mf.zip](https://github.com/user-attachments/files/18388619/Orca.Stanford.Bunny.3mf.zip)

[image]

Noisyfox, 2025-01-12:
This is normal behavior, see https://help.prusa3d.com/article/wipe-tower_125010
[image]

GlauTechCo, 2025-01-12:
> This is normal behavior, see

ohh i didn't know that, thanks for the info

closed 2025-01-12 by GlauTechCo

LehoooJ, 2025-01-23:
issue remains in

[image]

[image]

Noisyfox, 2025-01-23:
> issue remains in
> 
> [image]
> 
> [image]

That's not a bug

tome9111991, 2025-06-05:
Is there a way to skip last ramming move?