OrcaSlicer/OrcaSlicer #7986: The bridge layer before solid infill is missing when the "Detect narrow internal solid infill" option is enabled.

rosmolovsky, 2025-01-09:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Bambu Lab P1S

### How to reproduce

1. Open attached project file
2. Slice the object
3. Go to layer 55, 56 and you will see the missing bridges layer in the left and right narrow side areas of the object

### Actual results

When the "Detect narrow internal solid infill" option is enabled, in the narrow areas, printing with a concentric pattern is performed directly on top of the sparse infill without an intermediate layer of bridges. This leads to print defects and makes it impossible to complete the print. Problematic areas are marked with the arrow on the screenshots.

Layer 55:
[image]

Layer 56:
[image]

The result:
[image]

### Expected results

Narrow areas should include an intermediate layer of bridges before top solid infill layers.

### Project file & Debug log uploads

[missing-bridges-narrow-infill.zip](https://github.com/user-attachments/files/18359743/missing-bridges-narrow-infill.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Noisyfox, 2025-01-10:
The issue is not the "Detect narrow internal solid infill" option, but rather why those two places are treated as solid internal infill, instead of bridges (like how the middle part works).

Noisyfox, 2025-01-10:
The issue is caused by https://github.com/SoftFever/OrcaSlicer/blob/e3155861258205f93048bb9c10c3fae511bd6b15/src/libslic3r/PrintObject.cpp#L2693
I have no idea why this shrink is necessary.

Noisyfox, 2025-01-10:
By reducing this value the bridges are placed properly
[image]

Noisyfox, 2025-01-10:
Though I'm not sure how to properly solve this problem. Reducing these values might cause some other unexpected affects and has to be properly analyzed and tested.

hotellonely, 2025-01-31:
@Noisyfox I think I can provide something interesting to investigate:

When "Filter Out Small Internal Bridges" are set to "No Filtering",
it still would not generate any bridge under this surface:

[image]

However if I change "Ensure Vertical Thickness" to "Moderate",

[image]
Now suddenly it works as intended.

@igiannakas maybe you might have some valuable insights!

igiannakas, 2025-01-31:
> [@Noisyfox](https://github.com/Noisyfox) I think I can provide something interesting to investigate:
>
> When "Filter Out Small Internal Bridges" are set to "No Filtering", it still would not generate [...]

What is your ensure vertical thickness in the first example?

hotellonely, 2025-01-31:
> What is your ensure vertical thickness in the first example?

Currently any other value would cause this issue (same in bambu though)

closed 2025-05-09 by github-actions[bot]