OrcaSlicer/OrcaSlicer #7990: Change how top solid infill interacts with protruding elements

The760, 2025-01-09:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

OrcaSlicer has a behavior that I personally believe severely messes with top solid infill. When there's a protrusion from a large top solid layer, such as text, Orca will create holes in the shape of said text after the bridge infill layer, pictured here: 
[image]
This results in an ugly top layer finish, where the majority of areas not around the text look fine, but the text areas clearly stand out for having gaps as seen here: 
[image]

For those wondering, I'm using Archimedean Chords because I personally believe it looks better on a circular object such as this model. This same issue also happens on Monotonic/Monotonic Line.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

The ideal solution would be to either stop this behavior entirely, or add some sort of toggle in the quality tab so the user can choose to disable it themselves. As an experiment I cut down the model to see if I could get the top solid layer printed how I would prefer, and I was able to achieve that here: 
[image]

This prints the top solid layer with no holes for the text, meaning it can print much more consistently with no gaps. The text would start printing on the next layer.

I had started a help thread in the Orca discord, and another user named "MxBrnr" said they recall that only Cura and SuperSlicer don't have this behavior. So take that info as you will.

### Describe alternatives you've considered

Before anyone asks, yes I messed with speeds/temps, line widths, flow calibration, PA calibration, etc. None of that fixed this.

kylek29, 2025-01-14:
It'd be great to have a toggle for this, I noticed a long time back and it can introduce some artifacts with text and certain details.

drewsearcy, 2025-01-18:
I really need this as well. Anything with text on a top later makes the layers leading up to the transition look horrible and take forever.