OrcaSlicer/OrcaSlicer #7991: crash transferring simple scalar values between filament profiles

pleasantone, 2025-01-09:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.3.0-dev nightly as of 2025-01-08

### Operating System (OS)

macOS

### OS Version

Sequoia 15.1.1

### Additional system information

n/a

### Printer

BambuLab X1C

### How to reproduce

1. Empty new project.
2. Select system profile for filament as active, no other settings.

[image]

3. Open A/B settings comparison
4. Open user profile for filament into left column
5. Open system profile for filament into right column
6. Select transfer checkbox active
7. Select only a few simple scalar items (bed temperature, filament density, retraction settings)

[image]

8. Click green transfer button
9. Collect crash logs

### Actual results

Crash.

### Expected results

Simple items should transfer, system preset has been modified so behave as expected. The fact it's a system preset is kind of a red-herring, seems to happen with any transfer.

### Project file & Debug log uploads

[debug_log.zip](https://github.com/user-attachments/files/18367389/debug_log.zip)
[Orca Transfer Apple Crash Log.txt](https://github.com/user-attachments/files/18367425/Orca.Transfer.Apple.Crash.Log.txt)
[image]
[image]

### Checklist of files to include

- [X] Log file
- [ ] Project file

henrivdr, 2025-01-12:
This happens on Ubuntu 24.04 as well.

pleasantone, 2025-01-12:
Update:

Crash is in Tab::cach_config_diff, the config pointer being passed in is 0x58 and it's because in Tab::transfer_options, preset_from is NULL. At line 5509 we're dereferencing without checking for null.
Suggest changing 5509 from

```
cache_config_diff(options, &preset_from->config)
```

to
```
cache_config_diff(options, preset_from ? &preset_from->config : NULL);
```

However I'm not going to submit a PR because I'm not sure if that's a proper fix or a patch because `m_presets->find_preset(name_from)` should not be failing?

[image]

pleasantone, 2025-02-12:
Still a problem

m67hoff, 2025-02-15:
Transfer between filament profiles doesn't work at all   ( 2.3. 0-dev ccdba78 from Feb. 15  - Mac OS 15.3.1)
BTW: It worked/works with latest stable 2.2 

[image]

luksto, 2025-03-04:
Have the same Problem on Arch

evilDave, 2025-03-06:
Still seeing this issue on beta2 macos. There are probably at least 5 other duplicate reports (all with "I have searched the existing issues" checked - looks like that rule works...)

Probably duplicates:
https://github.com/SoftFever/OrcaSlicer/issues/8632
https://github.com/SoftFever/OrcaSlicer/issues/8674
https://github.com/SoftFever/OrcaSlicer/issues/8678
https://github.com/SoftFever/OrcaSlicer/issues/8715
https://github.com/SoftFever/OrcaSlicer/issues/8732

Noisyfox, 2025-03-07:
Will take a look

m67hoff, 2025-03-07:
Does this mean  I can/should verify with a fresh nightly build tomorrow ?

Noisyfox, 2025-03-07:
> Does this mean I can/should verify with a fresh nightly build tomorrow ?

You could download the PR build as soon as it finished.

PR #8744 (2025-03-07): Fix preset transfer crash
The issue is caused by #7573, the order of the combox no longer matches the order of enum `Preset::Type`, which breaks `DiffPresetDialog::get_left_preset_name` and `DiffPresetDialog::get_right_preset_name`, which rely on the combox order:
https://github.com/SoftFever/OrcaSlicer/blob/36a5a14507db6cdecbfc9b0d2768f06b18123f30/src/slic3r/GUI/UnsavedChangesDialog.cpp#L2384-L2394

This PR fixes this by using `std::find_if` instead so it no longer relies on the enum order.

Fix #7991
Fix #8632
Fix #8674
Fix #8678
Fix #8715
Fix #8732
closed 2025-03-07 by SoftFever