OrcaSlicer/OrcaSlicer #7992: Move post-test purge line from rear to front of plate

pleasantone, 2025-01-09:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I could have sworn I saw an open issue on this but it must have been on reddit because I can't find it now.
For the BambuLab P1/X1 printers, the machine start G code extrudes a purge line, runs a bunch of calibrations, and then does a second purge line at the back of the plate. BambuStudio changed this behavior in 2024, I would like us to follow suit, it's easier to scrape off purge lines without removing the plate from the bottom and this line parallels the original right next to it so it fits fine. It's been sized for X1/P1 plates.

### Which printers will be beneficial to this feature?

Marlin

### Describe the solution you'd like

https://github.com/bambulab/BambuStudio/commit/3e35fd82d50f80da500ab7d7f6c435b98e2ba966
https://github.com/bambulab/BambuStudio/commit/13ca5a05d8def3256c97ae6050e07124e96111f0

The following gcode matches bambulab's commits at the very end of machine gcode start, you can see the change by looking inside the if-statements.

```
M975 S1 ; turn on mech mode supression
G90
M83
T1000
G1 E{-retraction_length[initial_no_support_extruder]} F1800
{if true}
; if move purge line to front XXX
G1 X18.0 Y2.5 Z0.8 F18000.0;Move to start position
{else}
; original rear purge code
G1 X128.0 Y253.0 Z0.2 F24000.0;Move to start position
{endif}
G1 E{retraction_length[initial_no_support_extruder]} F1800
M109 S{nozzle_temperature_initial_layer[initial_no_support_extruder]}
{if true}
; new front purge line
G1 Z0.2
G0 X239 E15 F{outer_wall_volumetric_speed/(0.3*0.5)     * 60}
G0 Y12 E0.7 F{outer_wall_volumetric_speed/(0.3*0.5)/4   * 60}
{else}
; original rear purge code
G0 X253 E6.4 F{outer_wall_volumetric_speed/(0.3*0.6)    * 60}
G0 Y128 E6.4
G0 X252.5
G0 Y252.5 E6.4
G0 X128 E6.4
{endif}
```

Noisyfox, 2025-01-10:
Does P1 have this issue as well? I have a P1 and never see purge line at the rear of the plate.

Rayn2k18, 2025-04-20:
It would be a nice feature (moving purge line from rear to front as it is now - v2.3.0 -, created blobs and 1st layer failure on my x1-c, so a choice would be simpler)