OrcaSlicer/OrcaSlicer #7993: Internal wall not precise and unexpected

AMASTRO00, 2025-01-09:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

Windows 10 21H2 LTSC

### Printer

Creality k1C

### How to reproduce

simply print my project. Today no way to solve the issue

### Actual results

wall leak
[image]
[image]

### Expected results

linear circle wall
[image]

### Project file & Debug log uploads

[Test_Trafila_PASTAMAKER_Master Churros.zip](https://github.com/user-attachments/files/18368260/Test_Trafila_PASTAMAKER_Master.Churros.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Anything else?

In attach my project. try to print and you can see the result. I printed the same project in Cura to make sure it wasn't a printer problem and the result was perfect.

dfosantos, 2025-01-10:
How many did you print?

AMASTRO00, 2025-01-10:
at least 10 with different settings. always the same result.

AMASTRO00, 2025-04-11:
Sure. This issue is stale but has not been solved.

closed 2025-07-18 by github-actions[bot]