OrcaSlicer/OrcaSlicer #7997: Inner/outer/inner with 2 walls starts with outer instead of inner wall

tlhintoq, 2025-01-10:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.3-dev

### Operating System (OS)

Windows

### OS Version

Win10proX64

### Additional system information

[image]

### Printer

Any

### How to reproduce

1. New job
2. Add a cube
3. 2 walls
4. inner/outer/inner order
5. slice

### Actual results

Outer wall prints before inner

### Expected results

Inner/outer/inner
if there's no 3rd wall then still should be
Inner/outer/...

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [X] Log file
- [X] Project file

discip, 2025-01-10:
@tlhintoq
If you are using the latest dev build this might be related to #7951 and if so will be fixed with the merge of #7959. 😊

tlhintoq, 2025-01-10:
> @tlhintoq If you are using the latest dev build this might be related to #7951 and if so will be fixed with the merge of #7959. 😊

Except I don't use the Arachne engine (due to the problems in it like that one).
As seen in the screen capture this is with `Classic Wall Generator`

discip, 2025-01-10:
Ups. 😅
Sorry, I missed that crucial detail. 🤦‍♂️

buzzhuzz, 2025-01-10:
Sorry my ignorance, but how inner/outer/inner supposed to work with only 2 perimeters?

I've just looked into the feature tooltip and it is clearly says "this feature required minimum of 3 walls to be effictive..."

MxBrnr, 2025-01-11:
I would argue this is working exactly as intended. The whole purpose of `inner/outer/inner` wall order is so that the outer wall is printed before the immediately adjacent inner wall. In the case of a 2 wall print, you are simply eliminating the first inner wall, effectively changing this setting to `outer/inner`. As previously mentioned and detailed in the tooltip, `inner/outer/inner` only works with 3 or more walls.

tlhintoq, 2025-01-12:
> to be effictive...
'Effective' is not the same as `work in sequence as described`.  It may not be `as effective` with only 2 walls but it should still follow the same sequence.

@MxBrnr That's one way to look at it.

The other is... Its called `inner/outer/inner`.
That's the order it should print in.
1 - inner
2 - outer
3 - inner

> The whole purpose of inner/outer/inner wall order is so that the outer wall is printed before the immediately adjacent inner wall.
One interpretation.
The other is:
"The inner wall should print before the outer so there is something for the outer wall to stick to on slopes/overhangs."

Here's the common workflow...
I'm debating between 2 walls and three... Wall order of `(1)inner/(2)outer/(3)inner`
I slice it at 3... And I get an inner wall before the outer wall.  _"Ok, I like that.  But I don't need 3 for this object."_
So I reduce it from 3 walls to 2.  That should be all that needs to be done, and I should still get the `(1)inner` wall before the `(2)outer` wall because there is no `(3)inner` l to make

henrivdr, 2025-01-12:
I agree 100% with @tlhintoq

staal54a, 2025-01-18:
@tlhintoq My understanding is that inner/outer/inner gives more dimensionally accurate parts because in the somewhat standard/default inner/outer setting, the outer wall slightly overlaps the adjacent inner wall and is pushed out slightly. So generally, the reason to use inner/outer/inner is more dimensionally accurate parts (plus some claim it makes it look better, I haven't noticed much of a difference myself).

Outer/inner should also be more dimensionally accurate than just inner/outer so from the mindset of someone that chose inner/outer/inner for more accuracy, outer/inner does make sense when using 2 walls.

I can see why you would want to prefer inner/outer with 2 walls (better overhangs for example) but I think the test case you've presented is somewhat flawed. If you are forcing 2 walls for a whole part and you want to force inner/outer, you can just select that option. It's an extra click or two and takes more time, but there is a relatively easy way to get the intended behavior with your provided scenario.

I think a better scenario is if you slice with 3 or more walls but have thin sections of the part that only have two walls then you hit this "issue". However, my preference for this case, if using inner/outer/inner, would still be to do outer/inner since I would prefer the higher dimensional accuracy. Frankly, if it worked the way you described here, I would look at that as a bug myself. And there is a tooltip that explicitly calls out using a minimum of 3 layers. Maybe the wording could be stronger (I personally think it's fine), but using software in a way that is not intended and clearly spelled out by documentation shouldn't be a bug.

Maybe what this should be is a feature request to add some additional capability to allow users to choose what the behavior would be if a part/section would have 2 walls. Could just be two separate options like "Inner/Outer/Inner (Inner Last)", which is the current behavior, and then "Inner/Outer/Inner (Outer Second)" which would be the behavior you are requesting. I could see value in something like this where the 2 wall case allows usage of the "Precise Wall" feature.

henrivdr, 2025-01-19:
As a happy medium.  Maybe when there are thin sections of a 3+ wall part that only have two walls, the slicer progress bar should show a warning/caution.

MxBrnr, 2025-01-22:
"The inner wall should print before the outer so there is something for the outer wall to stick to on slopes/overhangs."

If it did this, it would be called `inner/outer`, which already exists. This is simply not what the feature is for, not how it is programmed to work. Again, this feature only works with 3 or more walls, so if you are using 2 or less, the order it prints in is a really a moot point.

tlhintoq, 2025-01-23:
> Again, this feature only works with 3 or more walls,

No.  It doesn't.  And therein lies the basis of this bug.  This feature/choice remains active when you only have 2 walls selected.  Its a perfectly legal choice to have inner/outer/inner selected and 2 walls.  So no, this feature does NOT only work when you have 3 walls.

If the feature did only work with 3 or more walls you'd get a popup:   

> `Inner/outer/inner` requires a minimum of 3 perimeters therefore your wall order choice has been automatically changed to `inner/out` to match your number of walls

And honestly, I'd be okay with that because it would mean the program was doing something for me, to help me maintain my choices.  I wanted inner before outer... I changed wall count... and the program did what it had to do to maintain my choice of inner before outer.  That would be nice.  That would be acceptable.  That would be expected and helpful.

But it doesn't.  `Inner/outer/inner` for 3... should still be `inner/outer` if I drop the wall count to 2.  Its not rocket surgery to maintain wall order.  Its not user friendly for dropping wall count to ALSO change wall order especially when in plain English its still  `inner/outer/inner` there's just no 3d perimeter to take that 3rd `inner`.

You could totally drop the 2 wall version of `inner/outer` if `inner/outer/inner` did the right thing by doing `inner/outer` when the wall count was two.

staal54a, 2025-01-24:
@tlhintoq There is a difference between `works` and `enabled`. It can remain enabled when slicing with only 2 perimeters but not work/function correctly. So the tooltip does imply that, although I could see that the language could be updated to explicitly state that it will not function as intended when slicing with less than 3 perimeters. If the nature of your bug was simply to update that tooltip, I could see that (though I think the point still comes across that using `inner/outer/inner` with less than 3 walls shouldn't be done).

I also think you are focusing on the wrong use case. If a user is changing the number of walls for a whole print, I don't think it's unreasonable to levy the responsibility onto the user to have to update some additional settings to reflect the new print settings, including the wall order. A much more reasonable use case to target is a print that you set with 3 perimeters but has thin sections that only print 2 perimeters. For this particular scenario, you would still want `inner/outer/inner` for the 3+ perimeter section but there are sections that you can't explicitly set to `inner/outer`.

However, what you are suggesting I personally don't agree with at all and I actually prefer the current implementation. As I stated before, `outer/inner` advertises more dimensional accuracy than `inner/outer`. I use `inner/outer/inner` for dimensional accuracy purposes so if I was printing with 2 walls (thin part or changing the wall number for some reason), I would still prefer `outer/inner` to maintain my goal of dimensional accuracy. 

The other part I think you may not be grasping onto is that the most important part of `inner/outer/inner` is the fact that the **outer wall is printed _before_ the adjacent inner wall** which gives it the dimensional accuracy. If you follow that, then the current ordering when using two walls would make sense, because it would still be printing the **outer wall _before_ the adjacent inner wall** (it would just be that there is no non-adjacent inner wall to print first). This is the main reason I don't think this is a bug because it behaves exactly as the developer intended.

My suggestion would be, unless your only goal is updated tooltip language or a new name, that this be moved to a feature request instead of a bug. Under that you could submit some different ideas like allowing there to be a secondary option to specify the 2 wall behavior when using `inner/outer/inner` or maybe having a popup warn the user when changing the wall count to 2 (though this doesn't really help the thin part case though). I could easily see the value add of allowing users to basically configure the slicer to use `inner/outer/inner` 3+ walls but then setting a different ordering for 2 walls, especially if precise wall could also be applied.

MxBrnr, 2025-02-01:
> No. It doesn't. And therein lies the basis of this bug. This feature/choice remains active when you only have 2 walls selected. Its a perfectly legal choice to have inner/outer/inner selected and 2 walls. So no, this feature does NOT only work when you have 3 walls.

It's literally right there in the tooltip:

[image]

closed 2025-05-10 by github-actions[bot]