OrcaSlicer/OrcaSlicer #7998: Customize Build Plates

RikshaDriver, 2025-01-10:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Currently the list of Build Plates is based off Bambu Lab defined ones. It's an "all or nothing" scenario where we must use all the build plates or disable custom build plates.

Printers from other manufacturers may have variations to the build plates which are not customizable.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Allow Build Plate customization for a given printer... such as:

Option A:
- Limit Build Plates for existing list
- Allow string renaming of existing build plates for non-Bambu Lab printers

Option B:
- Option A &
- Allow custom build plates for non-Bambu Lab printers

kylek29, 2025-01-14:
Yeah, it'd be nice to add the functionality as custom build plates get more ubiquitous. 

Other settings that could be added:

- Build Plate Artwork File
- Z-Offset Value
- AruCo ID (for printers that look for a QR code)

jjenkUW, 2025-06-30:
This is a bugbear for me too and I would like a solution.