OrcaSlicer/OrcaSlicer #8004: I just updated the Orca-Flashforge for V1.3.0 and lost ALL filament profiles and ALL the processes

Drakkengard, 2025-01-10:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

Orca-Flashforge V1.3.0

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

Flashforge 5M Pro

### How to reproduce

Today all mine Flashforge 5M Pro showed a new update and recommended install the new version of the Orca-Flashforge. So I did it and all my custom material presets and all the processes disappeard. I ran a 3d printing business and these printers are the majority of my farm. I have some old data saved but they don't save after reload.
I really need those profiles and processes. 
What can I do?

### Actual results

[image]

### Expected results

Filament profile and processes appears like always

### Project file & Debug log uploads

nope

### Checklist of files to include

- [ ] Log file
- [ ] Project file

MxBrnr, 2025-01-10:
You will have to post this issue to Flashforge and request tech assistance from them. It is a separately maintained software that is only _based_ on OrcaSlicer, and the devs here do not have any control over it.

avavdoshin, 2025-01-11:
You should change for your filament profiles "inherits" part to current.
In previous versions base filament profiles has names like "Flashforge PLA @FF AD5M", in new version it was renamed to something like "Flashforge PLA"
Pla silk was renamed from PLA-SILK to PLA Silk, etc.
Also for process files you should rename from "inherits" part from f.,e. "@FF AD5M 0.4 Nozzle" to "@Flashforge AD5M 0.4 Nozzle"

jcforbes57, 2025-01-11:
Best I can tell is to email support@flashforge.com which is what I'm going to do about this issue and several others with the new version. So that this comes up in more google searches (how I ended up here), my other issues I'm going to email about with the new version:
-Crashes at 65% when generating g-code using tree supports set to "tree slim" style
-Ignores file name format settings entirely

inkeliz, 2025-01-11:
I think this issue must be closed, since it's not related to Orca. It's exclusively related to Orca-Flashforge, which is a fork violating the OrcaSlicer license, and can partially be accessed at https://github.com/FlashForge/Orca-Flashforge.

closed 2025-04-20 by github-actions[bot]