OrcaSlicer/OrcaSlicer #8011: XY hole compensation causes print artefacts on large holes

laukejas, 2025-01-11:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.3.0-dev

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

Voron 2.4

### How to reproduce

When a hole in the model has a partial opening to the outside of the part, and xy_hole_compensation is set to a non-zero value, OrcaSlicer assumes the layers with the opening to be part of the external perimeter (which is technically true), which results in print artifact - a ridge inside the hole:

[image]
[image]
[image]

While this is technically correct, it is clearly undesirable in such cases, and there is no way to fix it, at least none that I can find.

To recreate the issue:

1. Open the attached .3mf.
2. Slice the model with the current settings (default profile + xy_hole_compensation set to 0.15mm)

### Actual results

A ridge forms in the large hole on the layers where it connects to the external perimeter via a small hole.

### Expected results

The large hole should maintain it's diameter throughout it's entire height.

### Project file & Debug log uploads

[A00001091.zip](https://github.com/user-attachments/files/18387212/A00001091.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

Please make Orca aware of such situations, or at least provide the user with a way to fix these cases with some manual effort.

Mario-vminion, 2025-01-12:
it's an old bug and it returns....

laukejas, 2025-01-12:
@Mario-vminion So it is indeed classified as a bug? When I discovered it's due to hole connecting to the external perimeter, I kind of thought it's logical for software to do this, even if undesirable for the user. Do you mean to say it was solved somehow, but now returned?

Mario-vminion, 2025-01-13:
I had it reported in October 2023
[image]

laukejas, 2025-01-13:
@Mario-vminion Do you have a link to that issue? Not sure what you named it. Maybe we should merge them.

Mario-vminion, 2025-01-13:
No it was closed and it was solved…then

Mario-vminion, 2025-01-13:
maybe this one #736

RobertKoval, 2025-01-20:
Annoying issue. Up.

Mario-vminion, 2025-04-21:
Is this solved yet

closed 2025-07-28 by github-actions[bot]