OrcaSlicer/OrcaSlicer #8012: Wall artifacts on clean STLs

laukejas, 2025-01-11:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.3.0-dev

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

Voron 2.4

### How to reproduce

With some STL files, OrcaSlicer produces weird horizontal artifacts on some of the walls:

[image]

These artifacts are very much visible with Arachne (above), but with Classic, they are reduced - yet still there (below):

[image]

Normally one would suspect that the artifacts are in the STL, however, analyzing the STL shows no triangle lines/ends on the location of these artifacts. There are many lines there, but all of them vertical, and since STL triangles are flat, it means that these horizontal indentations cannot possibly exist in the STL.

[image]

This suggests that the STL is fine, but something happens during the slicing process. This is made evident by the Arachne making these artifacts far worse. Making the STL more coarse in CAD software seems to reduce these artifacts somewhat, which indicates that it is still at least partially related to the STL quality. The exact cause is unknown though. I suspect it has to do something with the perimeter lines crossing through so many triangles in such a short distance. Such high STL resolution is required for other model features though, making the use of coarser STL impractical.

To recreate the issue:
1. Open the provided .3mf
2. Slice the model

### Actual results

The wall beneath the thin rib should be smooth, as in the STL.

### Expected results

The wall has strange horizontal artifacts.

### Project file & Debug log uploads

[A00001091.zip](https://github.com/user-attachments/files/18387249/A00001091.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

discip, 2025-01-13:
This is due to the fact that the apparently flat surface has a very shallow curvature.

laukejas, 2025-01-13:
@discip Yes, but that is intentional. These artifacts are definitely not. Any way to fix this, or is it a bug?

discip, 2025-01-13:
I don't remember the exact cause, but I it has something to do with the rounding of calculation values.
The curvature is shallow enough to sometimes make the slicer 'think' it is a straight line, while in other cases it interprets it as a curve.

discip, 2025-01-13:
I don't remember the exact cause, but I think it has something to do with the rounding of calculation values.
The curvature is shallow enough to sometimes make the slicer 'think' it is a straight line, while in other cases it interprets it as a curve.

discip, 2025-01-13:
The curvature is shallow enough to sometimes make the slicer 'think' it is a straight line, while in other cases it interprets it as a curve.

laukejas, 2025-01-13:
@discip Understood, thanks for the explanation. Any way to fix it? This curvature is required, as well as a high quality STL. Are there any Orca settings that I could change that would fix this without hurting print quality?

discip, 2025-01-13:
I've tried it myself in the past but haven't found a solution yet.

If I remember correctly, the root cause is much deeper in the core.
I think the culprit is OpenCASCADE.

laukejas, 2025-01-13:
@discip Understood. Let's hope devs notice this and could comment further.

laukejas, 2025-02-14:
@discip  Any updates on this? I confirmed that these artifacts appear on printed parts as well, not just in the slicer. They really mess up the geometry when trying to print molds...

closed 2025-05-24 by github-actions[bot]