OrcaSlicer/OrcaSlicer #8018: Outer perimeter overhangs are treated as bridge, with "Slow down for curled perimeters" enabled

Boerny41, 2025-01-12:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Line is detected as "overhang wall"
[image]

Overhang speed is set to 100mm/s, but the line still uses 50 mm/s as if it is a bridge.
[image]

As far as I understand, "Slow down for curled perimeters" should prevent the above behavior and force it to use the overhang speed settings. Below is another screenshot with this option disabled, however the line uses the bridge speed in both cases
[image]

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Use overhang speed for overhang perimeters when "slow down for curled perimeters" is enabled

### Describe alternatives you've considered

Change the line type to bridge. It is very confusing when the line type menu says it is an overhang, but it ignores the overhang settings and instead uses the bridge settings. Even more so when the option that should force overhang walls to use overhang speeds does not do so in this specific case.

MxBrnr, 2025-01-12:
What version of OrcaSlicer are you using?

Boerny41, 2025-01-12:
2.2.0 latest stable

Noisyfox, 2025-01-14:
Could you provide a test 3mf?

Noisyfox, 2025-01-14:
I guess it has something to do with this line
https://github.com/SoftFever/OrcaSlicer/blob/1088d0a6c713250ee3d14da791e27b76b4339363/src/libslic3r/GCode.cpp#L5219-L5220

ShantanuNair, 2025-04-04:
@Boerny41 There is a bug in the core speed setting logic which doesn't incorporate flow ratios for bridges/overhangs/ perimeters etc. I made a PR here: https://github.com/SoftFever/OrcaSlicer/pull/9210#pullrequestreview-2741664201 which may address your issue, so if you have the bandwidth, test with my changes, or test if you don't see the issue when using classic overhang mode, since that wouldn't use the erroneous code path.