OrcaSlicer/OrcaSlicer #8019: Nightly: Changing printer preset loses all transferred process settings

jeversol, 2025-01-12:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.3.0-dev cd4a1e0

### Operating System (OS)

macOS, Windows

### OS Version

MacOS 15.2, Windows 11 Pro 24H2 26100.1742

### Additional system information

Mac: Mac mini, M2 Pro, 16GB RAM.
PC: AMD Ryzen 5 5600X3D, 32GB RAM, nVidia GeForce RTX 4070

### Printer

Bambu Lab P1S

### How to reproduce

- Open a project that has customized process settings (infill density, supports, etc)
- Note the customized settings
- Change the printer profile to another existing printer
- On the "Transfer or discard changes" window, choose Transfer
- Notice in the UI, the Process tabs with modifications are orange, indicating settings are modified
- Change to another process tab 

### Actual results

- Note all of the orange tab titles revert to default color
- Note customized settings in the process profile are now the defaults

### Expected results

- Modified settings should have persisted through the printer profile change when "Transfer" is selected for transferring the modified values

### Project file & Debug log uploads

[Duskscale-Snake_v2.1_SilkPLA.3mf.zip](https://github.com/user-attachments/files/18390359/Duskscale-Snake_v2.1_SilkPLA.3mf.zip)
[Screen Capture.mov.zip](https://github.com/user-attachments/files/18390361/Screen.Capture.mov.zip)
[debug_Sun_Jan_12_12_21_14_5400.log.0.zip](https://github.com/user-attachments/files/18390364/debug_Sun_Jan_12_12_21_14_5400.log.0.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

I included the debug log (nothing useful), the original project file, and a screen capture showing the behavior. Same behavior occurs on Windows nightly. There is a [similar issue](https://github.com/SoftFever/OrcaSlicer/issues/1635) reported in 2023 but the referenced link to the fix is dead.

hotellonely, 2025-01-13:
[image]

Can confirm, after transfer all the settings are lost.

Noisyfox, 2025-01-13:
The issue is caused by #7860

PR #8027 (2025-01-22): Fix extruder number issue when SEMM is not enabled
Redo the fix for #7860 and #7483.
Also fix #8019
closed 2025-01-22 by Noisyfox