OrcaSlicer/OrcaSlicer #8023: Arachne breaks the wall to chunks when there is a thin part

eternal-sorrow, 2025-01-13:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Linux

### OS Version

Gentoo

### Additional system information

CPU: AMD Ryzen 7 PRO 5850U with Radeon Graphics
RAM: 32GB
GPU: Advanced Micro Devices, Inc. [AMD/ATI] Cezanne [Radeon Vega Series / Radeon Vega Mobile Series] (rev d1)

### Printer

Elegoo Neptune 4 Max

### How to reproduce

1. Slice a model with a thin area where thickness of a mode is only about one wall width, but said thin area has thicker areas on both sides.
2. Make sure you use Arachne wall generator.

### Actual results

Arachne splits the wall into separate chunks, printing a thin wall chunk separately from other outer walls. It doesn't try to make the wall continuous:
1.
[image]
2.
[image]
3.
[image]

This results in two normal seams in the corners and two "seams' between the wall chunks.

### Expected results

Something like this:
[image]

### Project file & Debug log uploads

[wall.zip](https://github.com/user-attachments/files/18393254/wall.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

discip, 2025-01-13:
This is no bug, this is physics. 😊
The nozzle you are using is 0.5 mm in diameter, the wall you are trying to print in 2 adjacent lines is only 0.4 mm. So each of those lines would need to be a mere 0.2 mm in width.

How do you think is this going to happen? 

You either need to change the nozzle to a 0.2 mm one, or increase the wall thickness to at least 0.8 mm.

eternal-sorrow, 2025-01-13:
My nozzle is 0.4mm, but I don't want to print it in 2 adjacent lines, just one.

eternal-sorrow, 2025-01-13:
I see how the diagram that I drew might be confusing. Let me clarify it:
[image]
Circle marks the start of the wall, square - the end.

discip, 2025-01-13:
If you designed the part yourself I recommend to increase the wall thickness.

discip, 2025-01-13:
If you designed the part yourself, I recommend increasing the wall thickness.

eternal-sorrow, 2025-01-13:
Impossible in this case. The wall is supposed to be this thin. I originally wanted to make it even thinner - 0.2mm and print it in another orientation to use the layer thickness as a limiting factor, but it turned out 0.4mm is acceptable.

hotellonely, 2025-01-16:
> Impossible in this case. The wall is supposed to be this thin. I originally wanted to make it even thinner - 0.2mm and print it in another orientation to use the layer thickness as a limiting factor, but it turned out 0.4mm is acceptable.

you need a smaller nozzle,

eternal-sorrow, 2025-04-17:
Automated issue closing shouldn't be a thing.