OrcaSlicer/OrcaSlicer #8030: Mint Linux unable to to get past startup wizard, cannot action GET STARTED button

fenixryan, 2025-01-13:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Linux

### OS Version

Linux Mint 22 Cinnamon

### Additional system information

System:
  Kernel: 6.8.0-51-generic arch: x86_64 bits: 64 compiler: gcc v: 13.3.0 clocksource: tsc
  Desktop: Cinnamon v: 6.2.9 tk: GTK v: 3.24.41 wm: Muffin v: 6.2.0 vt: 7 dm: LightDM v: 1.30.0
    Distro: Linux Mint 22 Wilma base: Ubuntu 24.04 noble
Machine:
  Type: Vmware System: VMware product: VMware Virtual Platform v: N/A serial: <superuser required>
    Chassis: No Enclosure type: 1 serial: <superuser required>
  Mobo: Intel model: 440BX Desktop Reference Platform serial: <superuser required>
    uuid: <superuser required> BIOS: Phoenix v: 6.00 date: 11/12/2020
CPU:
  Info: 4x 1-core model: 11th Gen Intel Core i7-11850H bits: 64 type: SMP smt: <unsupported>
    arch: Tiger Lake rev: 1 cache: L1: 4x 80 KiB (320 KiB) L2: 4x 1.2 MiB (5 MiB)
    L3: 4x 24 MiB (96 MiB)
  Speed (MHz): avg: 2496 min/max: N/A cores: 1: 2496 2: 2496 3: 2496 4: 2496 bogomips: 19968
  Flags: avx avx2 lm nx pae sse sse2 sse3 sse4_1 sse4_2 ssse3
Graphics:
  Device-1: VMware SVGA II Adapter driver: vmwgfx v: 2.20.0.0 ports: active: Virtual-1
    empty: Virtual-2, Virtual-3, Virtual-4, Virtual-5, Virtual-6, Virtual-7, Virtual-8
    bus-ID: 00:0f.0 chip-ID: 15ad:0405 class-ID: 0300
  Display: x11 server: X.Org v: 21.1.11 with: Xwayland v: 23.2.6 driver: X: loaded: vmware
    unloaded: fbdev,modesetting,vesa dri: vmwgfx gpu: vmwgfx display-ID: :0 screens: 1
  Screen-1: 0 s-res: 1920x1080 s-dpi: 96 s-size: 508x285mm (20.00x11.22") s-diag: 582mm (22.93")
  Monitor-1: Virtual-1 mapped: Virtual1 res: 1920x1080 hz: 60 size: N/A modes: max: 1920x1080
    min: 640x480
  API: EGL v: 1.5 platforms: device: 0 egl: 1.4 drv: vmwgfx device: 1 drv: swrast gbm: egl: 1.4
    drv: vmwgfx surfaceless: egl: 1.4 drv: vmwgfx x11: egl: 1.4 drv: vmwgfx inactive: wayland
  API: OpenGL v: 4.5 compat-v: 4.3 vendor: vmware mesa v: 24.0.9-0ubuntu0.3 glx-v: 1.4
    direct-render: yes renderer: SVGA3D; build: RELEASE; LLVM; device-ID: 15ad:0405
Audio:
  Device-1: Ensoniq ES1371/ES1373 / Creative Labs CT2518 driver: snd_ens1371 v: kernel
    bus-ID: 02:02.0 chip-ID: 1274:1371 class-ID: 0401
  API: ALSA v: k6.8.0-51-generic status: kernel-api
  Server-1: PipeWire v: 1.0.5 status: active with: 1: pipewire-pulse status: active
    2: wireplumber status: active 3: pipewire-alsa type: plugin
Network:
  Device-1: Intel 82371AB/EB/MB PIIX4 ACPI vendor: VMware Virtual Machine type: network bridge
    driver: N/A port: N/A bus-ID: 00:07.3 chip-ID: 8086:7113 class-ID: 0680
  Device-2: Intel 82545EM Gigabit Ethernet vendor: VMware PRO/1000 MT Single Port driver: e1000
    v: kernel port: 2000 bus-ID: 02:01.0 chip-ID: 8086:100f class-ID: 0200
  IF: ens33 state: up speed: 1000 Mbps duplex: full mac: <filter>
Drives:
  Local Storage: total: 100 GiB used: 14.46 GiB (14.5%)
  ID-1: /dev/sda vendor: VMware model: Virtual S size: 100 GiB tech: N/A serial: N/A fw-rev: 1.0
    scheme: GPT
Partition:
  ID-1: / size: 97.38 GiB used: 14.45 GiB (14.8%) fs: ext4 dev: /dev/sda3
  ID-2: /boot/efi size: 512 MiB used: 6.1 MiB (1.2%) fs: vfat dev: /dev/sda2
Swap:
  ID-1: swap-1 type: file size: 2 GiB used: 0 KiB (0.0%) priority: -2 file: /swapfile
USB:
  Hub-1: 1-0:1 info: full speed or root hub ports: 6 rev: 2.0 speed: 480 Mb/s lanes: 1
    chip-ID: 1d6b:0002 class-ID: 0900
[... 297 lines cut ...]
(orca-slicer:5113): Gtk-CRITICAL **: 21:39:07.218: gtk_window_resize: assertion 'height > 0' failed

(orca-slicer:5113): Gtk-CRITICAL **: 21:39:07.219: gtk_window_resize: assertion 'height > 0' failed

(orca-slicer:5113): Gtk-CRITICAL **: 21:39:07.227: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:5113): Gtk-CRITICAL **: 21:39:07.424: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:5113): Gtk-CRITICAL **: 21:39:08.441: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:5113): Gtk-CRITICAL **: 21:39:08.441: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:5113): Gtk-CRITICAL **: 21:39:08.443: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:5113): Gtk-CRITICAL **: 21:39:08.443: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:5113): Gtk-CRITICAL **: 21:39:08.444: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:5113): Gtk-CRITICAL **: 21:39:08.447: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:5113): Gtk-CRITICAL **: 21:39:08.447: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

** (orca-slicer:5113): CRITICAL **: 21:39:08.453: Cannot register URI scheme wxfs more than once

** (orca-slicer:5113): CRITICAL **: 21:39:08.453: Cannot register URI scheme memory more than once

bazza14, 2025-01-17:
Getting the same issue on Fedora. Tried 2.2 and 2.3 app images then the 2.3 flatpack. The flatpack worked for a while until i tried to connect to the printer now it launches for a short while and then crashes. In all cases the logs are full of gtk errors

louhy, 2025-02-03:
Maybe confirming some things that DO work (for me) will help narrow down a pattern with these issues.

I **don't** have this problem with Mint 21.1. But I'm not running in a VM like the OP, and I'm using KDE on top of XFCE. Also using an AppImage like the OP (no Flatpak), but I confess I did build my own AppImage from source (I doubt that matters). And running an AMD GPU and a build of Orca 2.3.0.

_More importantly, I get those same Gtk-CRITICAL (supposedly) "errors" in my log._ So I don't think those logs are directly related to your problem(s).

Also you don't need to run the AppImage with sudo.

@Crono141 Your issue sounds different, probably you need a new issue report or need to go search for one that sounds like your own? The GTK-Criticals do not seem to be so critical. I've seen them in other applications too.

Crono141, 2025-02-03:
My issue is indeed different, and related to Nvidia hating Wayland. Switching back to x11 works around my problem.

bazza14, 2025-02-04:
> My issue is indeed different, and related to Nvidia hating Wayland. Switching back to x11 works around my problem.

My issue was the same as yours, thanks for posting your solution. Switched to X11 and the flatpack version is working.

serjepatoff, 2025-02-28:
This dirty hack helped me:
$ WEBKIT_DISABLE_COMPOSITING_MODE=1 /Path/To/AppImage

closed 2025-06-07 by github-actions[bot]