OrcaSlicer/OrcaSlicer #8031: No Sparse layer and  extra prime tower options dsabled for Bambulab A1 Printers

Jakub1177, 2025-01-13:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

Prusa MK3S / Bambulab A1

### How to reproduce

1 Multicolor , 2 Enable prime tower , 3 No sparse layer options not visible

### Actual results

BBL printers have no option to use "no sparse layer" option and other features as the other printers can.

Prusa
[image]
[image]

BBL A1
[image]
[image]

Extra features  are actually working on Creality or Prusa and other printers so it can be disabled on puropse.
I would like to know. 

No Sparse layer option should be "safe" to use on bed slingers if you make sure the clereance of extruder and X axis is OK.

On the other hand it should be used with caution on CoreXY printers when printing a tall object which can later during the print poke to the top of the printer. I could see it is it enabled for Creality K1 for example and allow me to use no sparse layer option even on object 240mm tall ( potential new feature to prevent a crash on printers with closed top and disable it from certain height.)
[image]

Thank you for your support.

### Expected results

Extra options of prime tower to be enabled same as for the other printers like Creality or Prusa.

### Project file & Debug log uploads

[Prusa_no sparse layer_OK.zip](https://github.com/user-attachments/files/18402695/Prusa_no.sparse.layer_OK.zip)
[BBL_A1_no sparse layer_NOK.zip](https://github.com/user-attachments/files/18402702/BBL_A1_no.sparse.layer_NOK.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Noisyfox, 2025-01-14:
BBL printers has different prime tower logic where those options don't apply.

Jakub1177, 2025-02-06:
@Noisyfox  Would you mind to change the label to an enhancement 
I understand the prime tower and purging out old filament is different but the requirement to print full tower when timelapse is disabled is quite useless.

closed 2025-05-16 by github-actions[bot]