OrcaSlicer/OrcaSlicer #8043: Spiral vase

boinappi, 2025-01-14:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.3.0-dev

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

RatRig VC3.1 enclosed

### How to reproduce

1. Import a Spiral Vase Mode compatible model (e.g. https://www.printables.com/model/379782-gnome-vase-mode)
2. Enable "Spiral Vase" in "Others->Special Modes" (global)
3. Set "Bottom shell layers" to 0 in "Strength Top/bottom shells" (global) to print this without bottom
4. Give it a little more bed adhesion by adding an inner brim in "Others". Set "brim type" to "inner brim only".
5. Hit slice

### Actual results

no inner brim

### Expected results

inner brim

### Project file & Debug log uploads

[Gnome_vase_80mm_brim.zip](https://github.com/user-attachments/files/18414739/Gnome_vase_80mm_brim.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Anything else?

It works with outer brim. So i guess it's meant to work with inner brim also.

aufflick, 2025-01-27:
Just noticed the same, also fwiw if you select inner+outer brim you do indeed get an inner brim (but with the not desired outer brim obv).

closed 2025-05-05 by github-actions[bot]

zsoltvaradi-git, 2026-11-09:
Can we please re-open this???? I am having the same issue. When bottom layer is set to 0 but there is infill, there should be inner brim!!!

zsoltvaradi-git, 2026-11-09:
[image]
There should be inner brim around all the infill lines that touch the base board.