OrcaSlicer/OrcaSlicer #8045: Z-axis height is not reporting correctly

jryan15, 2025-01-14:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

windows 11

### Additional system information

dell laptop amd ryzen 5000 series, amd radeon graphics

### Printer

Bamby P1S

### How to reproduce

1.  Open attached file and click on the thin Ohio state graphic.
2.  See that it is an assembly with 4 layered objects
3.  Click on any of the that the thin 0.6mm layers black, gray, white 
4.  Choose the "Move" function
5.  Notice that the z-axis position is shown as way off the build plate (>25mm) even though the parts are sitting near zero
6.  Caveat, this assembly on plate 2 was once on plate 1 which was configured as print by object
7.  Still, moving the black, gray, white layers to be stacked perfectly on top of the red cube is terrible without good z-axis positions

[image]

### Actual results

The height of the black, gray, white layers shows as ~25mm.

### Expected results

The height of the black, gray, white layers should be around 0.5mm not ~25mm.

### Project file & Debug log uploads

Please see the attached zip.

[ohio state_example.zip](https://github.com/user-attachments/files/18416384/ohio.state_example.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Anything else?

I have several complex assemblies where I try to stack or specifically place features in the z-axis and sometimes need to setup[ specific layer height modifiers for color blending.  Often with complex assemblies with many components and lots of build plates.  It is so much trouble to figure out where objects truly are in the z-axis.  I mean so hard...  And I'm a practicing engineer.  I try hard to understand the z-axis mystery.

tlhintoq, 2025-01-15:
That's how the coordinate system in Orca works.  
Z is not measured off the bottom of the object.  All values are from the center of the object in all axis.

[image]

There's a number of tickets requesting that be changed to the lower|left corner like every other program.
[image]

kylek29, 2025-01-15:
I do think it might be time to consider moving to a Per-Object coordinate system (object assembly/models reference world, components/modifiers reference object). The world approach made sense before the Per-Object and all of the downstream modifiers were added.

closed 2025-04-24 by github-actions[bot]