OrcaSlicer/OrcaSlicer #8046: Sacrificial Tower for minimum layer time

galaxyman7, 2025-01-14:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I am working on tuning ironing settings, and finding that the minimum layer time is very important to let the part fully cool to a steady state temperature so the top surface is flat when it is ironed. But if I slow down the print speed of the part, it becomes overheated and overextruded due to the nozzle being in contact with the part for longer, and the extrusion rate being slightly higher when printing slower. Even with a high fan speed, this messes with how the top layers turn out.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

So what I'm thinking is adding one of two features. Both would work for this purpose. It just depends on how easy it is to implement:
1. Add a per object checkbox setting that allows you to disable slowing down that object for the minimum layer time. This way you can make a sacrificial tower manually, uncheck that box, and let it be sped up and slowed down instead of your object.

OR

2. In the cooling options for the filament, add a checkbox for "sacrificial cooling tower". This would disable slowing down any objects for the minimum layer time, and add something like a wipe tower that would be slowed instead. 

This would have a very positive effect on print quality for many parts. It would also be great for layer adhesion, where you don't want to use too much fan because it will weaken a part. 

### Describe alternatives you've considered

I have tried making a separate tower and slowing the print time for the whole tower until it meets the minimum layer time. This works, but it slows down the print immensely if you have small features like the top of the benchy which require another 20 seconds or so to cool off enough.

### Additional context

I have been experimenting with 3d printers for around 12 years, and I am a mechanical engineer by trade.

VigilantGardener, 2025-01-15:
Would the integration of something like this achieve what you want?
https://youtu.be/P6Y8uUPd3yg?si=Qqb7HZ_AEEpST0l0

I'm not trying to argue with this enhancement request, but with what you described it made me think of that video and the temperature control script it's addressing.

galaxyman7, 2025-01-15:
@VigilantGardener that does somewhat address the issue, but you still end up with less layer adhesion that way, since some of your layers will be printed colder than others. For the best consistency of strength, you would need to print everything at the same temperature and speed. It makes it a lot easier to tune that way too. And I think it would be far easier to implement.

The variable temperature is a pretty complex solution that would take a while to tune, and is kind of a "black box" solution so it's hard to troubleshoot if it is causing issues. That said, it is very cool and would be great to implement also.

hkkxx, 2025-06-25:
I think it makes sense, but if the fan speed and chamber temperature can have a dynamic smooth transition, it will solve the problem better.

soyuka, 2026-08-29:
## RFC: implement "park & wait" — with an answer to the standing objection, real-world data, and an incremental plan

I'd like to revive this with (1) measured data from a real failure case, (2) a direct answer
to the objection that killed #4191, and (3) an implementation approach that reuses three
mechanisms already in the tree. I'm asking for a **go/no-go in principle** before writing code.

### The case slow-down cannot solve

Printing a Tr16×4 threaded rod, 134 mm tall, standing, 0.6 nozzle, 0.15 mm layers (PLA).
Layer area ≈ 2 cm². With `slow_down_for_layer_cooling` active, the print fails: slowing down
keeps the 220 °C nozzle hovering over the same 2 cm² *longer* — radiant heat from the heater
block softens the thread crests, and the continuous mechanical contact excites wobble on a
slender column. Slow-down makes both problems *worse*, structurally: what this geometry needs
is for the nozzle to **physically leave** between layers.

### The recommended workaround works — here is its invoice

In #4191 the suggestion was to add a sacrificial object next to the part. We did exactly that
(Ø12 tower, same height, ~29 mm away, as a separate object): the rod printed perfectly on the
first try — clean crests, no wobble marks, threads mate correctly with their printed nut.

The invoice for that workaround, per print:
- **12 g of PLA thrown away** (a hollow tube variant still costs ~4 g);
- the tower must be **modeled per part** (matched height, safe distance) — a manual step every time;
- it defeats slicer time estimation and wastes bed space.

The workaround being effective is precisely the argument *for* the feature: park & wait is
that same nozzle-detour, minus the wasted plastic, computed by the slicer instead of hand-built
by the user. (We've since reused it on a second slender part — Ø10 pin topping out at 100 mm —
with a hollow-tube variant of the tower: same result, and still ~4 g of plastic in the bin.)

### Answering the ooze objection directly

@igiannakas's concern (in #4191) is legitimate: an idle hot nozzle oozes, and the return can
leave a pressure-starved seam. Four lines of defense, in order of effectiveness — the first
one already exists in this codebase:

1. **Standby temperature drop while parked** (`OozePrevention` / `standby_temperature_delta`
   is already in-tree for multi-extruder idling): −30/−50 °C drops viscosity-driven ooze by an
   order of magnitude. Proposed setting `park_and_wait_standby_delta`, default 0 (off).
2. **Full retract before the park** (standard `GCodeWriter::retract()` — same as Cura).
3. **The park point is far from the part** — chosen by `TimelapsePosPicker`, which already
   knows how to avoid objects, the wipe tower, and the rod. Whatever drips, drips there. This
   is the same bet "traditional" timelapse mode has been making for years.
4. **Priming on return**: v1 does nothing special (standard unretract, documented risk); if
   measurement shows it matters, a later PR can park on the prime tower when one exists —
   which is literally the "add a cube" suggestion, automated.

Honest position: ooze is not *eliminated*, it is *relocated and reduced*. The question is not
"is there ooze" but "does ooze at a far park point cost more than 12 g of sacrificial tower
per part and a nozzle hovering over the part". That question is answerable by test prints,
and I'm volunteering to run them.

### Why Cura's implementation is not the model

Cura has `cool_lift_head`, but (checked in `LayerPlan.cpp`) it only retracts and z-hops
**3 mm, hardcoded, with no XY move** — the nozzle keeps radiating onto the part from
point-blank range (their #3919 complains about exactly this). The XY park is the part that
matters for slender columns.

### Implementation sketch (details available)

- **Scope: filament settings**, next to the existing cooling block — same scope as
  `slow_down_layer_time` and `dont_slow_down_outer_wall` (#5148 established that the minimum
  layer time behavior is amendable per-filament for a quality motive):
  `park_and_wait_for_layer_cooling` (default **off**), `park_and_wait_min_time`,
  `park_and_wait_max_time`, `park_and_wait_standby_delta`.
- **Architecture**: geometry and timing live in different pipeline stages, so `GCode.cpp`
  emits a geometrically-correct, state-balanced park block wrapped in sentinel comments
  (the existing `;_EXTRUDE_SET_SPEED` pattern), and `CoolingBuffer` — which is the only place
  that knows the real layer time — either substitutes the actual dwell or deletes the block.
  No duplicated time estimator, no fixed park point.
- Known traps already mapped: `CoolingBuffer` counts `G4` into layer time (must exclude the
  sentinel block from its own accounting); Klipper needs `M400` before `G4`; the block must
  sit outside `EXCLUDE_OBJECT` labels; multi-object layers already aggregate correctly at the
  `cooling_buffer_flush` boundary (to be locked by a test).
- **Incremental PRs**: (1) extract the existing dwell abstraction — trivial, standalone value;
  (2) *warn* when minimum layer time cannot be met (the ask of #12258 — indisputable, and it
  puts the gap on record); (3) the core park & wait; (4) multi-object test hardening;
  (5) anti-ooze refinements if measurements call for them.

I have a full plan with file:line references and test strategy (existing
`tests/fff_print/test_cooling.cpp` helpers) ready to share. Before any PR: is there
maintainer appetite for this behind a default-off filament setting, or is "add a cube"
considered the final answer?

ikcalB, 2026-09-01:
I'd be curious, to see how your proposed changes will work out (oozing, pressure inside nozzle when starting next layer) in comparison to "add a cube"

Warning & possible automatically adding a sacrificial object (with minimum filament weight wasted, i.e. tower, b/c that code already exists in orca) would be on my wish list, and could facilitate adoption of the other features?  
(if the other proposed features are [strictly] opt-in and orca defaults to the sacrificial object strategy, then there should be no downside to these changes what so ever)