OrcaSlicer/OrcaSlicer #8051: Make "Change Filament" More Accessible

weaseljohnson, 2025-01-15:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

The "Change Filament" menu (changing which filament roll/slot is applied to a model) is currently located at the very bottom of the right-click menu.

It is a feature that I use often since I keep multiple spools in my AMS and have different settings for different materials. 

You can also change it in the objects tab of the process section, but that requires extra clicks beyond just right-clicking on an object.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

It would be nice if it was located near the top of the right-click menu.

tlhintoq, 2025-01-15:
Lower than the view window.  ??
The OS should intercept that and cause the context menu (right-click menu) to paint higher up so it is complete.  That's normal Windows behavior.  Is the program window painting beyond (lower) than the actual bounds of your display?
[image]

If it helps in the mean time...
You can right-click and change filament from the object list, too.
No need to move around the plates to bring an object in to view, at all.
[image]

weaseljohnson, 2025-01-15:
@tlhintoq I guess I was not clear in my description. For context, I'm using a mac, not windows. And the right-click menu does not get painted so that some of it is hanging off the screen, it gets truncated and the menu becomes scrollable.

As noted in my description, I was aware of the ability to change it in the objects list, but that requires several more clicks beyond just right-clicking in while in the global settings view.

tkunchick, 2025-02-04:
I just checked on my Mac, and works fine.

If using a scroll mouse, just scroll the wheel.

If using a touch pad, just hover over the dropdown arrow (⌄).

[image]

yw4z, 2025-02-10:
added to my list. maybe i can make menu much more compact with creating few sub menus

weaseljohnson, 2025-02-11:
@yw4z  yeah, the main goal in my mind would just be to make that menu option more readily accessible (i.e. not at the bottom of the right click menu)

yw4z, 2025-04-12:
i have tried to simplify menu. but it requires so much changes on code side since bottom menu items are dynamic and top menu items are fixed

[image]

@Noisyfox @SoftFever let me know your thoughts on this one