OrcaSlicer/OrcaSlicer #8055: Enable "Long retraction when cut" setting for A1/A1 mini

noiob, 2025-01-15:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

There could always be less waste when multi-material printing.

### Which printers will be beneficial to this feature?

Others

### Describe the solution you'd like

Now that it's supported in the firmware, I'd like to have the "Long retraction when cut" checkbox when printing on my A1 mini as well.

### Describe alternatives you've considered

The Alternative is using Bambu Studio I guess? It's available there.,

killerline10168, 2025-01-16:
+1
It feels like a no-brainer when the devs get some time for it.

yzheka, 2025-01-18:
I can see the button flashing for a moment and then disappear. So i think this will be available after orca sync code with latest bambu studio.

fabi986, 2025-02-02:
+1 i see the button but no matter what oI do cant press it

yzheka, 2025-04-09:
[image]

Looks like it is already fixed

noiob, 2025-04-09:
it is, closing

closed 2025-04-09 by noiob