OrcaSlicer/OrcaSlicer #8058: Walls being generated when 0 selected

headamame, 2025-01-15:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

Windows 10 and 11

### Additional system information

multiple windows based systems

### Printer

Bambu P1P

### How to reproduce

Open project file, click slice. 
Despite some bodies being selected to have 0 walls, walls are being generated.

### Actual results

[image]

### Expected results

[image]

### Project file & Debug log uploads

[conesClosedBack_Orca - bug report.zip](https://github.com/user-attachments/files/18428107/conesClosedBack_Orca.-.bug.report.zip)
[debug_Wed_Jan_15_09_05_28_30596.log.zip](https://github.com/user-attachments/files/18428709/debug_Wed_Jan_15_09_05_28_30596.log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

For some reason, when importing the equivalent Bambu project file into orca slicer, this error does not happen. Both original project files are attached below:

[OrcavBambuProject files.zip](https://github.com/user-attachments/files/18428754/OrcavBambuProject.files.zip)

Smiodar, 2025-01-16:
For Voron it sliced correctly once I switched the printer to Voron from P1P. I'm on the 2.3.0-dev, not 2.2.0. I have the same issue when slicing for P1P though.

herm, 2025-02-07:
I could solve this by switching from Arachne to classic wall generator.

closed 2025-05-16 by github-actions[bot]