OrcaSlicer/OrcaSlicer #8059: FLSUN SR Printer System profile has incorrect values under Motion ability

zbunston, 2025-01-16:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I have had an issue when I print a file sliced from Orca Slicer.  My printer runs extremely slow during and any print job after until I reset my printer to default.  I discovered today that under motion ability Emit limits to G-code is checked, and the values under speed limitation are incorrect.  X300, Y300, Z20, E30. This is a delta printer and the values in the firmware are X200 Y200 Z200 E100. After correcting this everything works fine again. 

### Which printers will be beneficial to this feature?

Marlin

### Describe the solution you'd like

Update the system profile to correct this by either unchecking Emit Limits by default or change the Speed limitation Values to match the machine firmware.

### Describe alternatives you've considered

Uncheck Emit Limits to G-code by default or correct the Values under Speed Limitations

### Additional context

I am not sure if this is the correct path to get this fixed, but have seen many other users on Facebook with the same issue and no solutions until someone mentioned today that it was changing their max feedrate

zbunston, 2025-01-16:
I just noticed that the default Max Acceleration for Z is also set incorrectly by default. It should match X and Y for a Delta if I am looking at this correctly. Same with Jerk Z-

manjaroblack, 2025-02-22:
> I just noticed that the default Max Acceleration for Z is also set incorrectly by default. It should match X and Y for a Delta if I am looking at this correctly. Same with Jerk Z-

The acceleration, speed, and jerk of the Z axis should not be equal to the X and Y axis. This is not referring to the delta motion but rather the cartesian motion. The Z speed, acceleration, and jerk should be set low as the movement is usually against gravity. 

I can’t speak for you, but for me running Klipper with input shaper tuned, 300mm/s is well within reason for the SR.