OrcaSlicer/OrcaSlicer #8063: FW 1.08.03.00 from Bambu WILL BREAK ORCASLICER for X, P and A series

WimVDK, 2025-01-16:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

Bambu is going to release a "security" firmware update that will essentiall break the current networking plugin, ALL functionality. 

https://blog.bambulab.com/firmware-update-introducing-new-authorization-control-system-2/

This includes

* initiating a print job (cloud AND LAN MODE!)
* calibrations
* AMS settings
* ....

> Network Plugin for Third-party Slicer
> 
> Network plugin API for Third-party slicing tools (e.g. OrcaSlicer) based on open-source Studio development will no longer be able to utilize Studio’s network plugin API for authorization control. For these users, [Bambu Connect client software](https://wiki.bambulab.com/en/software/bambu-connect?ref=blog.bambulab.com) will act as a replacement. This new software removes slicing functions while enabling remote control and print initiation.

Another piece of proprietary blob "Bambu Connect" WILL BE REQUIRED

https://wiki.bambulab.com/en/software/bambu-connect?ref=blog.bambulab.com and will need to be called using an url scheme

ANY AND ALL BAMBU USERS should immediately complain to Bambu Lab as this virtually takes full control over what YOU can do with YOUR PRINTER! DO NOT UPDATE TO 1.08.03.00 if you rely (like myself) on orcaslicer until support for Bambu Connect has been added!

### Which printers will be beneficial to this feature?

Others

### Describe the solution you'd like

Implement support for Bambu Connect

### Describe alternatives you've considered

There are no alternatives

24c, 2025-01-17:
Yeah, this is scandalous...just read it this morning. You will have to export .3mf files from Orca to Bambu Connect (beta, that doesn't support live çamera view still) so another layer of crap to go through. I really like that in Orca I can send GCode to more than one brand of printer I own.

WimVDK, 2025-01-17:
It will be able to be called via a formatted URL but it's still annoying that's for sure, they're taking control over OUR printer

ziehmon, 2025-01-17:
> Network plugin API for Third-party slicing tools (e.g. OrcaSlicer) based on open-source Studio development will no longer be able to utilize Studio’s network plugin API for authorization control. 

I am curious to see how they are planning to prevent the usage of libnetworking.so in external software (like Orca), since with BambuStudio you don't require BambuConnect, so the .so definitely includes the whole binding implementation.

This can be easily reverse engineered, same as the crappy encryption of log and config files. I've done it before and I will do it again.

ziehmon, 2025-01-17:
> Network plugin API for Third-party slicing tools (e.g. OrcaSlicer) based on open-source Studio development will no longer be able to utilize Studio’s network plugin API for authorization control.

I am curious to see how they are planning to prevent the usage of libnetworking.so in external software (like Orca), since with BambuStudio you don't require BambuConnect, so the .so definitely includes the whole binding implementation.

This can be easily reverse engineered, same as the crappy encryption of log and config files.

petercockroach, 2025-01-17:
Excuse my ignorance but shouldn't we use this report to just implement the new formatted URL scheme?

WimVDK, 2025-01-17:
There's no real answer to that but complain on social media, email them , file a complaint support ticket...

ziehmon, 2025-01-17:
> Excuse my ignorance but shouldn't we use this report to just implement the new formatted URL scheme?

This is one option, right. The other is to just build the functionality of the BambuConnect agent into Orca.

I had a quick check and basically, they started to use x509 for communication with the printer. (like mTLS). The certificate used for signing and encryption in the client likely will have to be trusted there (which they might rollout with the next firmware update).

Well, guess, where the certificate and it's key would have to be embedded: BambuConnect agent. It won't take long to reverse and obtain it. The "problem" Bambu Lab is trying to solve is rather complex and the abilities they have in "solving" it are limited.

But of course, one could follow their wishes and implement the URL handler.

ziehmon, 2025-01-17:
This is one option, right. The other is to just build the functionality of the BambuConnect agent into Orca.

I had a quick check and it looks like they started to use certificates (maybe x509) for message/payload encryption. The certificate used by the client (BambuStudio or BambuConnect) probably would have to be trusted there upfront (likely included in firmware update).

The thing is, the certificate and it's key would have to be embedded both in BambuStudio and BambuConnect. It won't take long to reverse and obtain it. The "problem" Bambu Lab is trying to solve is rather complex and the abilities they have in "solving" it are limited.

But of course, one could follow their wishes and implement the URL handler.

petercockroach, 2025-01-17:
> This is one option, right. The other is to just build the functionality of the BambuConnect agent into Orca.
> ...
> But of course, one could follow their wishes and implement the URL handler.

I think there's space for both solutions. I would suggest as a community we should prioritize the URL handler to get unblocked, then pursue the direct implementation. 

The problem is the latter may end up being a cat & mouse situation which isn't great for the community at large.

WimVDK, 2025-01-17:
Either way @SoftFever has his work cut out unless someone else implements support in a pull request. I know 1 thing I ain't updating fw unless I have too anymore

ziehmon, 2025-01-17:
> I think there's space for both solutions. I would suggest as a community we should prioritize the URL handler to get unblocked, then pursue the direct implementation.

I agree with you. 

> Either way @SoftFever has his work cut out unless someone else implements support in a pull request. I know 1 thing I ain't updating fw unless I have too anymore

Did anyone noticed that Bambu Lab states that @SoftFever was informed about this upront?

> Before announcing this change publicly, we shared it with the lead Orca Slicer developer to ensure alignment and collaboration. 

https://blog.bambulab.com/firmware-update-introducing-new-authorization-control-system-2/, section "FAQ"

mattgauf, 2025-01-17:
I had the same thought as @gschintgen, but on a more time sensitive level... The PR containing the updates to the Bambu material presets (and the fix for TPU for AMS) has been merged but Orca hasn't posted a release with those fixes. 

In this case I can manage, but I'm worried that future updates to material presets etc will become dependent on "playing nice" with whatever Bambu cooks up. So I definitely hope that @SoftFever will provide some clarity on their intentions for non-updated printers.

Anthony-Bec, 2025-01-17:
Bambu Lab released "bambu connect" which will fix this issue, you upload the .gcode.3mf file from orca and it will send it to the printer.

LBowers23, 2025-01-17:
Sure but will Orca be able to query the AMS for what filament is loaded so it knows how to slice and if Bambu introduces new filament types will Orca be able to recognize them?

WimVDK, 2025-01-17:
> Bambu Lab released "bambu connect" which will fix this issue, you upload the .gcode.3mf file from orca and it will send it to the printer.

It doesn't fix anything until OrcaSlicer is updated to automatically send sliced files through BC. And they still limit what YOU can do with YOUR printer even in kan mode

cryptonym64, 2025-01-18:
> Sure but will Orca be able to query the AMS for what filament is loaded so it knows how to slice and if Bambu introduces new filament types will Orca be able to recognize them?

Supposedly the read of values (other than can) is not protected.

cryptonym64, 2025-01-18:
> > Excuse my ignorance but shouldn't we use this report to just implement the new formatted URL scheme?
> 
> This is one option, right. The other is to just build the functionality of the BambuConnect agent into Orca.
> 
> I had a quick check and it looks like they started to use certificates (maybe x509) for message/payload encryption. The certificate used by the client (BambuStudio or BambuConnect) probably would have to be trusted there upfront (likely included in firmware update). Essentially, mTLS.
> 
> The thing is, the certificate and it's key would have to be embedded both in BambuStudio and BambuConnect. It won't take long to reverse and obtain it. The "problem" Bambu Lab is trying to solve is rather complex and the abilities they have in "solving" it are limited.
> 
> But of course, one could follow their wishes and implement the URL handler.

Would be interesting to know expiration dates on those certs as those effectively become the day the printer died if not renewed. A true end of life date.

ronoverdrive, 2025-01-18:
> This is one option, right. The other is to just build the functionality of the BambuConnect agent into Orca.
>
> I had a quick check and it looks like they started to use certificates (maybe x509) for [...]

The only problem with reversing it from their code is it becomes a cat and mouse game with every software update to reverse the cert again. Not to mention it may put Orca at odds with Bambu with them labeling us as some kind of cracking group they can issue C&D legal threats to. Honestly Bambu Connect wouldn't be so bad if it wasn't still in a feature incomplete Beta state and offered a real API that 3rd party slicers can connect through to maintain current functionality, but right now its half @$$ed and feels like they're trying to push us back to Bambu Studio.

In the mean time we don't seem to have a choice in the matter, implement a URL handler for printers on 1.08.xx+ and keep current functionality as legacy support for those of us who will stay on 1.07.xx. Until Bambu Connect becomes feature complete and obtains a Linux package I will be staying on 1.07.xx for the time being.

SoftFever, 2025-01-18:
Bambu informed me of this change two days before their announcement.
They provided access to a beta version of "Bumbu Connect." I gave them feedback to help improve the workflow if this becomes their chosen path. Nonetheless, sending a print through another software after slicing will always be inconvenient.  I requested an authorization key from them to enable OrcaSlicer to communicate with their device, just as BambuStudio does. However, I haven't received a response yet. I will share more updates once they become available. I'm uncertain how receptive they will be to authorizing OrcaSlicer with the same capabilities as BambuStudio. We'll have to wait and see.

Aggeloz, 2025-01-18:
> Bambu Lab released "bambu connect" which will fix this issue, you upload the .gcode.3mf file from orca and it will send it to the printer.

Why would we have to go through hoops just to send a file to the printer? It works nicely now, i dont need to add another program in the middle to take care of file sending, it's just going to make things slower.

WimVDK, 2025-01-18:
> Bambu informed me of this change two days before their announcement. They provided access to a beta version of "Bumbu Connect." I gave them feedback to help improve the workflow if this becomes [...]

Considering BS and OS are a fork of the other, can't you legally just reverse it?

mrambossek, 2025-01-18:
> Considering BS and OS are a fork of the other, can't you legally just reverse it?

the "main program" is open source. no need to RE anything. their binary blobs, that the main program interfaces with, are not. whether or not it is legal to RE that has nothing to do with OS' license.

SoftFever, 2025-01-18:
I heard back from their development team; they are not going to greenlight OrcaSlicer to send prints directly to their machine. It has to be done through their Bambu Connect application.

WimVDK, 2025-01-18:
Well that just sucks. Definitely not updating fw now untill they kill legacy mode

SoftFever, 2025-01-18:
> Considering BS and OS are a fork of the other, can't you legally just reverse it?

They use a closed-source network plugin to communicate with their printer. They are going to require authentication to use the plugin, so...

axhe, 2025-01-18:
@SoftFever Can you please keep the current functionality for all that go LAN Only mode and stop updates as it is now as a separate way of connecting addionally to whatever cloud connection they want to go for?

ziehmon, 2025-01-18:
If this was really about security and invalid costly usage of their APIs they would provide a key for usage of the networking plugin to Orca. They could literally review the usage/invocation in Orca and they'd have leverage to ensure its use properly.

This is NOT about any of the above. This is about lock-in and control. I am no pessimist but you have to face the facts.

All of you have correctly pointed out that obtaining certificates by reversing of the plugin is a cat and mouse game. But even if it's just about resisting their malicious practice, I believe it's the right choice to pick the battle.

psiberfunk, 2025-01-18:
> If this was really about security and invalid costly usage of their APIs they would provide a key for usage of the networking plugin to Orca. They could literally review the usage/invocation in Orca and they'd have leverage to ensure it is used properly there.
> 
> This is NOT about any of the above. This is about lock-in and control. I am no pessimist but you have to face the facts.
> 
> All of you have correctly pointed out that obtaining certificates by reversing of the plugin is a cat and mouse game. But even if it's just about resistance to their malicious practice, I believe it's the right choice to pick the battle. Edit: this however shall not be affiliated in any way with Orca due to various reasons, including but not limited to legal concerns.

What a shame it would be if someone patched their new DLLs/SO files and shared the patcher file, and a how-to-drop-it-in-the-right-place explainer....

ziehmon, 2025-01-18:
> What a shame it would be if someone patched their new DLLs/SO files and shared the patcher file, and a how-to-drop-it-in-the-right-place explainer....
> 
> The libraries have to run on untrusted computers... it's only a matter of time before someone cracks whatever obfuscation they put in their plugin.  This is security through obscurity.  Without a real root of trust they can't keep the secret keys secret.

What a shame, the minds of cybercrime gangs at GitHub always think alike!

hugheaves, 2025-01-18:
Yeah, the security argument for LAN mode makes no sense. The printer generates a secret, that is only available if you have physical access to the printer (i.e. shown on the printer's display), and will refuse any control operations without that secret. That is about as secure as you can get! What's really going on is that they don't want everyone to jump ship from "Internet mode" to "LAN mode" because LAN mode provides more flexibility. So, they're deliberately crippling LAN mode as well.

theDiverDK, 2025-01-18:
@SoftFever Did they give you any reason why they do this?

I wonder why they don't use a system like bearer tokens on webpages, or oauth2 to authenticate us, and allow programs to have access.

A lot of fortune500 companies use oauth2 and it's good enough for them.

To me it seems like they are either very bad at authentication/authorization understanding, or it's purely a management decision (and a really bad one). 
They should look at Sonos to see what happens when you make bad decisions. :-(

Let's hope, they wake up and understand how bad a decision it is.

SoftFever, 2025-01-18:
> [@SoftFever](https://github.com/SoftFever) Can you please keep the current functionality for all that go LAN Only mode and stop updates as it is now as a separate way of connecting addionally to whatever cloud connection they want to go for?

yeah, as long as the firmware is not upgraded, the existing plugin will still work

alea32, 2025-01-18:
I would suggest maybe people need to email the CEO with their complaints.  Apparently he can be reached at ferdinand.tao@bambulab.com

SoftFever, 2025-01-18:
> [@SoftFever](https://github.com/SoftFever) Did they give you any reason why they do this?
>
> I wonder why they don't use a system like bearer tokens on webpages, or oauth2 to authenticate us, and [...]

Reasons are not important IMHO :(

rocket59, 2025-01-18:
@alea32. I suspect they've got staff monitoring github/reddit etc. 

Better to just stop recommending theit printers to your friends. (and don't update your printer past 1.07)

psiberfunk, 2025-01-18:
> [@alea32](https://github.com/alea32). I suspect they've got staff monitoring github/reddit etc.
> 
> Better to just stop recommending theit printers to your friends. (and don't update your printer past 1.07)

Why not both? :)

xperiments, 2025-01-18:
Hi,

I am the maker of xtouch, 

As @SoftFever comments the only path by now is not UPDATE

If you check latest updates introduced really only minor changes from 1.05

From my point of view... V2 machines are coming....

So updates to V1 aside X1 will be minor...

PD: I have done a good move supporting Klipper LOOOL

OskarLindgren, 2025-01-18:
> There's no real answer to that but complain on social media, email them , file a complaint support ticket...
> 
> > Excuse my ignorance but shouldn't we use this report to just implement the new formatted URL scheme?
> 
> Yes, but that doesn't fix the inherit bad move by bambu. They have no authority to lock down EVEN LAN MODE!

I think one of the best things to do is point the bat signal to the sky and hope that EU consumer rights protections will save us. I've already sent in a complaint ticket - and I implore everyone to do the same. This frankly seems to be the best option at the moment

looxonline, 2025-01-18:
@SoftFever You have an opportunity to do something that may go down in the history books of legendary marketing hahaha...

[image]

Cypherous, 2025-01-18:
> I heard back from their development team; they are not going to greenlight OrcaSlicer to send prints directly to their machine. It has to be done through their Bambu Connect application.

Its a shame but its still workable, i'll keep using orca anyway so i'm not worried about connect being the go-between

SelfhostedPro, 2025-01-18:
do we know if this will impact aftermarket firmware like https://github.com/X1Plus/X1Plus ?

psiberfunk, 2025-01-18:
> do we know if this will impact aftermarket firmware like https://github.com/X1Plus/X1Plus ?

It will not.  X1Plus will not be adding this "feature".

Aggeloz, 2025-01-18:
> do we know if this will impact aftermarket firmware like https://github.com/X1Plus/X1Plus ?

I think the x1 plus replaces the bambu firmware completely so no. Also i hope their team can do something about the P and A series of printers, seeing where bambulab is heading...

SelfhostedPro, 2025-01-18:
Cool, I’ll throw a donation their way and recommend others do the same so we have a realistic alternative for hardware we already paid for. Thanks for the info.

WimVDK, 2025-01-18:
> > do we know if this will impact aftermarket firmware like https://github.com/X1Plus/X1Plus ?
> 
> I think the x1 plus replaces the bambu firmware completely so no. Also i hope their team can do something about the P and A series of printers, seeing where bambulab is heading...

You think SEVERELY WRONG. 'X1Plus' does NOT replace the original firmware. It's just a different GUI that sits on top of the original firmware, setup of x1plus downloads and unpacks the original firmware

Aggeloz, 2025-01-18:
> You think SEVERELY WRONG. 'X1Plus' does NOT replace the original firmware. It's just a different GUI that sits on top of the original firmware, setup of x1plus downloads and unpacks the original firmware

Yeah but they control which version of the original firmware gets uploaded, they can just choose to not move over to the newer firmware version.

odysseyofmayhem, 2025-01-18:
@Aggeloz the issue that X1Plus will face is that if the original firmware is removed by Bambu from their servers, the X1Plus can't build their firmware.  So the rug could be pulled on that project also.

To quote from the X1Plus README:
> The core concept of X1Plus is that we build an overlay on top of the Bambu Lab firmware, and replace only the parts that we need in order to launch X1Plus. We are very careful not to redistribute any of Bambu Lab's IP directly; when it's necessary to use or patch Bambu Lab binaries, we download them directly from Bambu Lab servers, and then patch them onboard the printer. We also try to be pretty careful to leave the printer in a "fail-safe" state by modifying as little of the on-printer flash as possible. Below is a rough flow of how X1Plus works, from installer through to normal boot.

Aggeloz, 2025-01-18:
> [@Aggeloz](https://github.com/Aggeloz) the issue that X1Plus will face is that if the original firmware is removed by Bambu from their servers, the X1Plus can't build their firmware. So the rug could be pulled on that project also.
> 
That's understandable but their work is valuable if someone decides to change the whole firmware and add klipper support or something like that.

cryptonym64, 2025-01-18:
> [@alea32](https://github.com/alea32). I suspect they've got staff monitoring github/reddit etc.
> 
> Better to just stop recommending theit printers to your friends. (and don't update your printer past 1.07)

I will actively post the status to any forum post asking "what printer to buy".  Let's work to take some money from their overstuffed wallets. Maybe they notice that.   I've been a huge fan and supporter of Bambu printers, but that ends now. Gloves off, ready for the show. May get banned from "official" but there are plenty of others plus YT vids. Leave comments everywhere.

cryptonym64, 2025-01-18:
> I would suggest maybe people need to email the CEO with their complaints. Apparently he can be reached at [ferdinand.tao@bambulab.com](mailto:ferdinand.tao@bambulab.com)

Email sent. My emails tend to p!ss people off but maybe he needed to hear it. 🤷‍♂️

mon5termatt, 2025-01-18:
Hot take. I dont mind this software as long as it keeps most of the process streamlines

Being able to pipe in more options, something like "?printerid=MYP1S&startnow=true" etc. would be very useful so that any third party software can just "start" a print but have a popup from the connect software that says (are you sure you would like to start this print)

odysseyofmayhem, 2025-01-19:
Reddit user hWuxH has decompiled/deobfuscated the Bambu Connect binary which is an Electron app.
I don't think this has the whole picture, but it shows how the signing is working and has at least some keys embedded:

https://hastebin.skyra.pw/pufugimoye.js

Gitouddahere, 2025-01-19:
Consensus seems to be that Firmware 1.07.XX or earlier is fine, and 1.08.XX is screwed. I currently have 1.08.02 in LAN only mode and everything works fine, and I don't see how that'll change without "upgrading" to the new evil firmware. Am I missing something?

buzzhuzz, 2025-01-19:
@Gitouddahere , keeping FW 1.08.02 and earlier will work for any current user of Bambu products. However, eventually all new devices will be shipped with 'evil' version, so we'll need to support such products in the future.

non-bin, 2025-01-19:
Is there enough time to push an update to OrcaSlicer with a notif warning people not to update??

buzzhuzz, 2025-01-19:
@non-bin , as far is I know, there are no functionality to push external notification to version of OrcaSlicer that have been already installed on the user machines. OrcaSlicer shall be updated by the user (or user shall agree to OrcaSlicer updates itself).

Anyways, there are interesting things provided by Bambu Labs in their ToS: see paragraphs 7.4 https://bambulab.com/en-us/policies/terms

> 7.4 Your Bambu Lab product will automatically search for and download new update packages to provide you with timely update services. These updates are designed to resolve cyber security loopholes and prevent new threats, and it is important to accept and install security related system updates in a timely manner. Due to the importance of these updates, your product may block new print job before the updates is installed, and will immediately provide update notifications to help you understand the related information.

SoftFever, 2025-01-19:
Based on what I've learned from the Bambu Connect Beta, this is the workflow with the their new firmware:

First, use OrcaSlicer (or any other non-Bambu application) to slice the model and save it as a gcode file. Then, invoke Bambu Connect with the saved gcode file, which OrcaSlicer can do programmatically. 
Next, the user must switch to Bambu Connect to complete the rest of the process to initiate the print:
1. Select which Bambu printer to use
2. Click the print button
3. Set the AMS mapping
4. Click the 'Send' button to start the print

Alaric323, 2025-01-19:
Speaking as someone with an X1 with x1plus installed: changes like these always upset me to the core. Subscription services seem to always be the cause of "why can't we have nice things", and is why more countries desperately need consumer protection laws as strong or stronger than the EU's.  Could this be a class action in Europe, and would the market share there be enough to make Bambu cave?

I saw that someone decompiled Bambu Connect and that it runs on Electron; does that mean it's possible to make a wrapper application for Connect where it automatically binds to Orca?  In the same way that you can do a Klipper connection and it takes over the Device tab, could a Bambu Connect wrapper be made to take over the Device tab and it works as normal?

It'd of course have to be a separately installed thing or installed in the same way as the network plugin is now; bundling Connect with Orca seems like the antithesis of open source.

tlhintoq, 2025-01-19:
Give them another 6 months - wait for all the yelling to calm down - wait for tools and scripts and whatever to get updated and all the newer users that didn't know what it was like in the "before times" to get into the eco-system.

THEN... the subscriptions will start.  No only are you... the owner... not an authorized user of your own $2k printer... but the next step will be to have to pay a subscription fee to keep using it.  Just wait and see.

Time for all these turn-key, know-nothing, never had to calibrate a printer in my life users to learn how to DIY with other brands of hardware.

ME: Mine are in LAN mode and will not be doing firmware updates.  They gave me good prints until now, no need to fix what's not broken.  But it does mean I'll never buy another one because it will come with the locked hardware.

[image]

Gridzilla, 2025-01-19:
> Then, invoke Bambu Connect with the saved gcode file, which OrcaSlicer can do programmatically.

Can't Bambu Connect be "bundled" in a separate tab like what is currently done for a web browser loading Mainsail for the Voron workflow? It's a deterioration compared to the current tightly integrated workflow for Bambu printers, but still somewhat integrated.

[187 more replies]