OrcaSlicer/OrcaSlicer #8068: seam gap path still wrong.

rocketscott, 2025-01-17:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.2

### Operating System (OS)

Windows

### OS Version

windows 11

### Printer

v3 ke

### How to reproduce

new install
orca tolerance test
seam gap 200%

### Actual results

path is missing a way point

[image]

### Expected results

perfect print first try

### Project file & Debug log uploads

[OrcaToleranceTest.zip](https://github.com/user-attachments/files/18451600/OrcaToleranceTest.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

@igiannakas ty for the overhang slowdown stuff.  i should be contributing more..

igiannakas, 2025-01-17:
Most likely this happens because your gap is too high pushing the seam far from the external perimeter in the corner. Try with an appropriate gap of around 10-15% and the wipe should be correct.

rocketscott, 2025-01-19:
10-15% wrong is less wrong than 200% wrong.  setting it to zero would be the only way to have a correct seam fill/wipe.

closed 2025-04-28 by github-actions[bot]