OrcaSlicer/OrcaSlicer #8082: velocity predetermination

zelk12, 2025-01-18:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

Nowadays the speed settings are changed to adjust the speed, but different speeds may be correct for different materials, which results in the need to manually change the speed settings each time depending on the material.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Add the ability to predefine speeds to the material's predefinition settings.

DVSVIDEO, 2025-01-18:
You set the speed for the material and then save it in a profile called PLAxxx 0,2mm_120_speed and then make another speed setting and save it as PLAyyy 0,2_110_speed. I think you are just starting 3D printing and don't have a good understanding of it....

zelk12, 2025-01-18:
There is such a very useful menu that allows you to change some of the parameters based on the material you are using, and a nice thing would be to add speed to that menu.
[image]

If we talk about Cura, you can choose which parameters depend on the material.
[image]

DVSVIDEO, 2025-01-19:
I don't understand why you want to assign a speed to a material? The speed depends slightly on the material, more on the shape of the model, the quality you want and so on... Why don't you make a process profile for different materials? You set all the data for that material and save it with the name of the material. Example: soft TPU is slow to print because of its elasticity. 1. I set the profile saved for the TPU in the process, which includes the very slow first layer with low acceleration and the other speeds with slightly higher acceleration. Accept that Orca is different from Cura and they will not reprogram it to get a Cura-2... There is a solution...don't be lazy... You can save a custom setting for each material, which can include everything, not just speed..

[image]

zelk12, 2025-01-19:
The main point of improvements is to optimise the process, not to repeat another object exactly, if there is an option in another slicer to make something more convenient, why not do it that way.
In this case the main problem is that besides speed there are many other parameters, which can often be similar in different materials, and as a result of experimentation, a more optimal variant of this parameter can be found. Now let's imagine we have 10 or 20 similar profiles, we have found a variant of supports that will work well on each profile. But as there are a lot of parameter profiles, it will be necessary to change each profile in manual mode, but the speed may depend on the material, in this case, we set the speeds in the material, but do not touch other parameters.

DVSVIDEO, 2025-01-19:
I understand what you're saying, but I've been using a 3D printer for 8 years (I rebuilt a simple model in years) and in my experience speed is much more dependent on the shape of the model (overhang, first layer) and the quality you want (should it be shiny, should it be matte, should it be a precise fit) than on the material. 
I go through all the settings before starting to print and adjust them to the model. I then also fine tune the speed to suit the subject. I also often change the loaded profile to match the object to be printed and therefore do not save the changes. It's just for this occasion. There is no setting or profile that is right for everything. That's why I find it unnecessary to assign speed to the material. 
I hope you understand me.

zelk12, 2025-01-19:
I realise this, and often change parameters as circumstances dictate, but it is not uncommon to find that certain speed parameters, for one material or another, may be appropriate for different objects. And when it is necessary to change several materials in a row, while the optimum parameters have already been found, it is a bit slow to work, then it is possible to make the possibility of auto selection of printing parameters depending on the material, through manual binding profiles.

haunter1982, 2025-01-24:
I also think the speed and even acceleration setting should be within the material settings. 
You can't print every material with the same speeds. And it doesn't make sense to have different material settings and then for every material setting an additional process setting.

What could be an "easy" way to achieve this slightly is to give a "max speed" and "max acceleration" "max jerk" setting inside the material profile. perhaps located where the max volumetric flow is.
So the ones who would like to set the speed for materials could set here the max speeds for their materials and then use the percentages in the process setting.

So for the ones who are not interested in the new way, nothing would change, and the one who would have the speed set for the filament could do it this way.