OrcaSlicer/OrcaSlicer #8085: Maximied Window Orca Slicer blocks Taskbar

Kingsidorak, 2025-01-18:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

10

### Printer

Bambu

### How to reproduce

Enable Hide Taskbar in Desktop Mode
Use Orca Slicer with the window maximized on the same monitor with a taskbar, and try to access the taskbar

### Actual results

Unable to access the task bar

### Expected results

The taskbar should come up like it does with all other programs

### Project file & Debug log uploads

.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

discip, 2025-01-18:
Unfortunately this behavior is also true for BambuStudio.

killerline10168, 2025-01-19:
+1 This behavior is very irritating, as I like to hide my taskbar due to my OLED monitor. 

@discip Do you mean it would make this issue more difficult to solve?

discip, 2025-01-20:
> Do you mean it would make this issue more difficult to solve?

I don't think so, it's just an observation.

PR #8118 (2025-01-22): Fix auto-hide taskbar overlapping issue when maximized
Fix #8085

Due to the limitation of windows, to allow the task bar to work properly, I have to leave a small gap at the edge of the window when maximized. This only take affect when auto-hide taskbar is enabled.

See https://stackoverflow.com/a/61540163/3289421

This code is adopted from https://gist.github.com/MortenChristiansen/6463580, huge thanks to @MortenChristiansen
closed 2025-01-22 by Noisyfox