OrcaSlicer/OrcaSlicer #8090: Nozzle collides -  "No sparse layers **on** "

lwbsbde, 2025-01-18:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

win11

### Printer

Custom DUAL Extruder Servo

### How to reproduce

1. Einstellung "Multimaterial"  -  Prime tower **on** - No sparse layers **on**

### Actual results

Nozzle move down before moving to the tower.
Nozzle vcollides with object

### Expected results

Nozzle move to the tower before move down

### Project file & Debug log uploads

[drehgriff.zip](https://github.com/user-attachments/files/18466159/drehgriff.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

[image]

Rneil-ca, 2025-01-19:
I am seeing this bug as well, when switching toolheads and staying on the same layer, the tool head is lowered before moving to the tower.  I have spent hours looking at this as thought I had something setup wrong, but I can not make this work.  It is lower the toolhead before it even starts the code for the prime tower or toolchange.

G1 X212.466 Y87.06 E.02475
; stop printing object sorting.stl id:0 copy 0
G1 E-.9 F2100
G1 Z.2 F24000 ;  <---Here is where it is lowering
M73 P34 R114
G1 E.9 F2100
;HEIGHT:0.2
;TYPE:Prime tower
;WIDTH:0.5
;--------------------
; CP TOOLCHANGE START
; toolchange #1
; material : PLA -> PLA
;--------------------
; WIPE_TOWER_START
M220 S100
; CP TOOLCHANGE UNLOAD

tomas-a-sanchez, 2025-01-22:
Hello people I decide also to try the feature... I was testing a benchy when suddenly I just heard the crack on the extruder.
here are some pictures of the problem...  @lwbsbde describes well the bug.

[image]

[image]

I am new to Gcode and some stuff about modding or coding etc... also new to orca slicer (which I found is better than the rest). How do I get a copy or extract of the gcode like @Rneil-ca did by pasting the g-code here ??? where I can get that from?

lwbsbde, 2025-01-22:
[image]

Save GCODE

Rneil-ca, 2025-01-23:
I created a quick post-processing script to temporally fix the issue.  I have attached it.  I have tested  it and works with my printer, but can not guarantee it will work with yours.  USE AT YOUR OWN RISK. You will need to rename the attached file from .txt to .py and add it to your post-processing scripts

[MultiCorrect.txt](https://github.com/user-attachments/files/18524488/MultiCorrect.txt)

tomas-a-sanchez, 2025-04-27:
> I created a quick python post-processing script to temporally fix the issue. I have attached it. I have tested it and works with my printer, but can not guarantee it will work with yours. USE AT YOUR OWN RISK. You will need to rename the attached file from .txt to .py and add it to your post-processing scripts
> 
> [MultiCorrect.txt](https://github.com/user-attachments/files/18524488/MultiCorrect.txt)

I have been trying to do that but nothing! at the end it was resolved on the newer version of orca slicer, but now I am using Snapmaker orca and I am facing the same thing! Cant make it work!

closed 2025-08-03 by github-actions[bot]

dragonmaus, 2025-08-17:
Just had this occur to me, using OrcaSlicer version 2.3.0, build 70931e5.

Printer got to the first filament change spot, then homed the nozzle in on the tower location, raised the bed, and only then moved to cut the filament. This, of course, rammed into my in-progress model and nullified several hours of printing.

dragonmaus, 2025-08-17:
@SoftFever

Not sure who to ping to get this issue reopened, since it was closed for no good reason but is still a problem.