OrcaSlicer/OrcaSlicer #8093: Temperature Settings For Bed Type Ignored, Always Using Smooth High Temp Settings.

efaden, 2025-01-19:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

macOS

### OS Version

15.2

### Additional system information

MacBook Pro M4 Pro

### Printer

Qidi Plus 4

### How to reproduce

With the default Qidi Plus 4 profile enable multiple bed type support.  Set the bed temperature for the filament under the "Textured PEI Bed" to something (say 10 degrees).  Set the Bed type to Textured PEI Plate.  Generate G-Code

### Actual results

Temperature used in the G-Code is the temperature for the Smooth High Temp Plate from the filament.

### Expected results

It should be using the temperature corresponding to the selected bed type.

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/18467336/log.zip)
[project.zip](https://github.com/user-attachments/files/18467337/project.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

SinisterMinister, 2025-01-30:
This is due to the incorrect variable used for the start gcode. Usages of `hot_plate_temp_initial_layer` should be `bed_temperature_initial_layer_single` instead. Update that in your starting gcode will fix this issue for you.

efaden, 2025-01-30:
How do you get the plate type to show up in preview?  Thanks!
-Eric

pm0u, 2025-02-08:
Is there any update needed for temp changes after the first layer? Thanks for the solution!

SinisterMinister, 2025-02-08:
I imagine that's handled in the generated gcode as that shouldn't be
printer specific. Is it not working? I haven't paid any attention to that
personally.

closed 2025-05-17 by github-actions[bot]