OrcaSlicer/OrcaSlicer #8097: Bambulab printer not connected after startup in LAN mode

schenlap, 2025-01-19:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.2

### Operating System (OS)

Linux

### OS Version

Debian trixie

### Printer

A1Mini/Bambulab

### How to reproduce

1. set LAN only Mode on printer
2. start orcaslicer
3. go device tab
4. -> Connecting...
5. tcp connected does not start with printer (can be seen in wireshark)
6. "Failed to connect to printer"
7.  click on the printer under My Device -> Printername (LAN)
8. I see tcp connection in wireshark is oppened an printer is listed as connected 

### Actual results

Printer does not get connected after startup. I see in wireshark that no connection is startet from orcaslicer. After clicking on "printernane (LAN)" the connection is started immediatly.

### Expected results

orcaslicer should connect on startup.

### Project file & Debug log uploads

[debug_Sun_Jan_19_10_44_12_18026.log](https://github.com/user-attachments/files/18468344/debug_Sun_Jan_19_10_44_12_18026.log)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

It is reproducible here and can be watched in wireshark.

3vi1, 2025-01-20:
Can confirm seeing same, built from current git master. 

It would be nice if it remembered that you had a LAN printer selected at shutdown and then automatically selected it at next startup.

Johnr24, 2025-01-20:
+1 on this, the ability to bind a printer to the slicer without having to type the code in each time would be lovely!

schenlap, 2025-01-22:
My original issue is not regarding forgetting access code. So I think it is not related to #6169.
Maybe it is related to network plugin, I do knot know. Everything I can debug I had done.
I only know that it does not work as expected, whatever reason it has. Because of #8063 more people will use LAN mode and more people will see it is not working as expected.

Noisyfox, 2025-01-31:
Please try this PR https://github.com/SoftFever/OrcaSlicer/pull/8256

schenlap, 2025-01-31:
Works now! Thank you for your work!

cryptonym64, 2025-02-02:
Sorry for the late reply, been working on Home Assistant updates and new
bug testing for them.  Lots of progress since Bambu went all power crazy.

Yes! this works as I had hoped now. Thank you!

closed 2025-02-05 by Noisyfox