OrcaSlicer/OrcaSlicer #8099: Add the ability to connect Bambu Lab printers in LAN Only Mode

Jwn5k, 2025-01-19:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

Bambu Lab is changing how third party apps talk to their printers over their cloud services for "security" reasons, It is a problem for every person with a Bambu Lab printer of any kind that uses their networked/cloud features as well as any third party slicer programs that talk to the printers.

### Which printers will be beneficial to this feature?

Others

### Describe the solution you'd like

Add the ability to bind a printer over the LAN network using the provided "LAN Only Access Code" in the format of "00000000" in any combination of letters and numbers, that the Bambu Lab printers have specifically for LAN only mode. Bambu Studio has this feature next to Bind With Pin when adding a new printer, and using a Pin needs to be connected to the internet to work, whereas the LAN only mode does not need to.

### Describe alternatives you've considered

1. Using Bambu Studio; This is the most practical solution as it has the capability built in
2. Using an SD card; This is not the most elegant solution, but it is the simplest, but not being able to use a fantastic feature on a product (wireless printing) is quite a downgrade in convenience, ESPECIALLY if one has multiple printers, such as in a print farm.

### Additional context

I am justifying this as means to keep Orca Slicer relevant to not have to go through yet ANOTHER translation layer as well as the Bambu Lab cloud service already due to the changes coming in firmware 01.08.03.00 (for X1 Carbon at least) that will outright block ANY app, including Orca Slicer, from accessing it's printers through their cloud services at all unless it uses another piece of software, Bambu Connect. This is a cludgy and non-user friendly thing Bambu Lab is doing, and I'd rather not give up Orca Slicer, as it has features that Bambu Lacks, such as great 3rd party printer support. This is a video on what Bambu Lab is changing: https://www.youtube.com/watch?v=aIyaDD8onIE as well as checking on this very GitHub repo seeing that it is a major issue for Orca Slicer! What I have done is I have blocked internet access for my X1 Carbon on my network so that it can't phone home and be prompted with said upcoming software update. I would rather lose out on cloud "features" for staying in more control for my own sake, then having Bambu Lab get what they want in this anti-consumer choice. Shame on Bambu Lab.

rocket59, 2025-01-19:
Also the ability to bind using IP please!

Flatfire, 2025-01-19:
Confused, as this option is plainly available in 2.2.0. 

[image]

[image]

Working as intended with my A1. Is this issue opened because you couldn't locate this feature? Or am I misunderstanding the nature of the problem?

With the printer in LAN mode, it can pair to OrcaSlicer with the code as expected, with no binding to an account required. Assuming you aren't on an X1 model printer with that beta firmware, you should be able to drop into LAN only and be okay, while blocking future firmware updates until (hopefully) Bambu eats enough flack to reverse course.

probonopd, 2025-01-19:
This is the issue:

[image]

viktorpenelski, 2025-01-19:
@Jwn5k I just paired my P1S to OrcaSlicer 2.2 using lan-only mode. What is the issue you are facing?

[image]

andornaut, 2025-01-19:
I'm facing the same issue.

I moved my p1p to a separate VLAN and separate subnet. Broadcast messages, which are presumably how OrcaSlicer discovers the p1p, do not work across these subnets. This is intentional.

"Discovering devices" in the way that OrcaSlicer does is a nice convenience, but it should also be possible to simply specify the IP address of the printer.

Flatfire, 2025-01-19:
Fair enough, that's a reasonable ask. There's no reason not to allow direct communication with the printer via a specified IP. 

If your firewall/switch hardware supports it, then allowing broadcasts between VLANs might be a workaround. If you're in a home environment with a more complex network such as PF/Opnsense, then there is a plugin that allows for relaying broadcast packets. Otherwise I think only Cisco supports it in enterprise environments.

andornaut, 2025-01-19:
Yep, I'm using pfsense with [avahi](https://avahi.org/) as a mDNS repeater. I can see `vahi-browse -art` other devices on this subnet, like my security cameras, but I do not see the bambulabp1p.

Perhaps, there's some other issue at play here. Any advice?

In any case, I can connect to the printer just fine via [this Home Assistant Integration](https://github.com/greghesp/ha-bambulab) by specifying the IP address of the p1p, but I cannot see the device in OrcaSlicer.

Is there some way to manually add the printer, such as via one of the configuration files in `~/.config/OrcaSlicer/`?

achtnullzwei, 2025-01-19:
It uses SSDP for discovery. I am trying to set up udpbroadcastrelay on pfsense to get it sorted but no luck so far. Need to tweak it more. 

Anyway, +1 for directly specifying the IP, or any other details that are otherwise transferred via the SSDP lookup, manually.

andornaut, 2025-01-19:
I actually got it working. I [wrote down the instructions](https://github.com/andornaut/til/blob/master/docs/3d-printing.md#device-discovery-across-subnets-using-pfsense) I followed in case it's helpful:

1. Install the [UDP Broadcast Relay package](https://docs.netgate.com/pfsense/en/latest/packages/udpbroadcastrelay.html)
2. Configure one "interface" to bind the two subnets for each of the following ports: 1900,1990,2021
   * According to [this thread](https://forum.bambulab.com/t/use-ssdp-standards/7173/2), the p1p uses udp:1900, which is standard, whereas the X1C uses udp:1990, which is non-standard
3. Configure firewall rules to allow traffic between subnets on ports 1900,1990,2021
   * Use the [firewall log](https://docs.netgate.com/pfsense/en/latest/monitoring/logs/firewall.html) (Status > System Logs > Firewall) to figure out what rules to create 

(See the [network ports](https://wiki.bambulab.com/en/general/printer-network-ports) wiki page.)

achtnullzwei, 2025-01-19:
Awesome thx. Interestingly enough, logs show 1990 and 2021 for me (p1s) when i tracked the printer in the fw logs. Will check your instructions tomorrow :)

andornaut, 2025-01-19:
> logs show 1990 and 2021 for me (p1s) 
Yeah, I'm seeing UDP traffic <bambulabp1p>:**1900** to 239.255.255.250:**1990**

1900 to **1990**. Weird, but oh well.

Lattyware, 2025-01-19:
> Yep, I'm using pfsense with [avahi](https://avahi.org/) as a mDNS repeater. When I run `avahi-browse -art`, I can see other devices on this subnet, such as my security cameras, but I do not see the bambulabp1p.
> 
> Perhaps, there's some other issue at play here. Any advice?
> 
> In any case, I can connect to the printer just fine via [this Home Assistant Integration](https://github.com/greghesp/ha-bambulab) by specifying the IP address of the p1p, but I cannot see the device in OrcaSlicer.
> 
> Is there some way to manually add the printer, such as via one of the configuration files in `~/.config/OrcaSlicer/`?

My manual work around for now is using [a script which just fakes the SSDP packets from the printer](https://gist.github.com/Alex-Schaefer/72a9e2491a42da2ef99fb87601955cc3), after that everything works fine.

I would like to add one other request: please don't lock this to literal IP addresses: support domains. I run local DNS for my network and use names for everything rather than remembering IPs, it would be nice to just be able to do that, sometimes software will over-validate inputs and demand literal IP addresses, when it's all going to standard libraries that would accept either.

Noisyfox, 2025-01-20:
Just to keep in mind, LAN only mode will break with the new firmware update even if you managed to bind the printer in the app, because sending any "critical" command to the printer requires authentication through BBL's plugin regardless, which blocks any 3rd party app other than BBS.

cryptonym64, 2025-01-20:
> Just to keep in mind, LAN only mode will break with the new firmware update even if you managed to bind the printer in the app, because sending any "critical" command to the printer requires authentication through BBL's plugin regardless, which blocks any 3rd party app other than BBS.
> 
> With that said, we will add subnet binding support by syncing with latest BBS code base, when time is available.

That's why you go to your router and disable internet access to the printer's  MAC address. No Handy App, no bambu slicer and no internet = usable printer.

Glod76, 2025-01-20:
I've been using the x1plus firmware overlay

Noisyfox, 2025-01-20:
> That's why you go to your router and disable internet access to the printer's MAC address. No Handy App, no bambu slicer and no internet = usable printer.

Only if you managed to stay on the old firmware. For any BBL printers ship with newer firmware that won't work.

StainlessSteele, 2025-01-20:
+1 to this. I have moved my printers to LAN only and would like to not log into any Bambu services, I have my printer on a separate network from my PC. How can we help enable this?

NebulaBC, 2025-01-21:
This would be a massive help for me. I recently purchased a Bambu Lab printer, and with news of the coming update I moved which network my printer is on, allowing me to block its access to the external internet. My main LAN has full access on the necessary ports, both ways, and both octoprint and home assistant are working just fine with it. I am just lacking a way to forward broadcast packets. My printer is essentially unusable for me outside of SD card printing until I can add it directly by IP in OrcaSlicer. I can maybe switch it to a generic octoprint printer, and copy over the gcode required for all the Bambu features, but that's an unnecessary pain.

It's surprising to me that in 2025 an open source program like this wouldn't support direct connections.

StainlessSteele, 2025-01-21:
> This would be a massive help for me. I recently purchased a Bambu Lab printer, and with news of the coming update I moved which network my printer is on, allowing me to block its access to the [...]

On my router, I enabled an IGMP Proxy between the two subnets and it instantly worked. It discovered and found the printer and I can send prints to it directly from a PC on the other subnet.

alaneb, 2025-01-21:
It looks like the A1 series and P1 series of printers give you a pin code option for binding, but this article on the bambu wiki states that the X1 series do not have the capability for a pin code, only an access code.  So while pin code is available in Orca right now, the access code is not an option to add the X1 series in LAN only mode.  It should be noted, the access code is not a 6 digit pin code, it's an 8 character code.

Wiki link:
https://wiki.bambulab.com/en/bambu-studio/manual/pin-code

Orca:

[image]

Bambu Studio:

[image]

StainlessSteele, 2025-01-21:
I think that article on X1 is old if you are in LAN mode you should be able to enable/view this on the X1 as well:
https://wiki.bambulab.com/en/knowledge-sharing/enable-lan-mode

alaneb, 2025-01-21:
It looks like the links we posted were both edited within a couple of months of each other last year, but the one you gave me shows what my X1C screen looks like.  I don't have the PIN code, only the access code, which Orca doesn't give me an option for.  If I try to add a printer in Orca, the only option is with PIN code, which looks like this.

[image]

I'd say it's expecting a 6 character or digit code, which the X1C doesn't show me.  I checked in each screen available on the printer's interface and there's no sign of a code that fits.  I'm not on the beta firmware, but I am on the most recent released version at this time which is 1.08.02.00.  According to Bambu's post regarding changes, they go live with 1.08.03.xx firmware which is still a beta release.  Unless I'm just missing something I don't see a way to add my X1C LAN only to Orca.  

I'd be happy to provide any extra info if it helps.

NebulaBC, 2025-01-21:
> On my router, I enabled an IGMP Proxy between the two subnets and it instantly worked. It discovered and found the printer and I can send prints to it directly from a PC on the other subnet.

I tried that on my OpenWRT router, no dice. Not sure if it's a configuration issue or what, but I've had no luck with anything I've tried to relay the mDNS.

cryptonym64, 2025-01-21:
> It looks like the links we posted were both edited within a couple of months of each other last year, but the one you gave me shows what my X1C screen looks like. I don't have the PIN code, only the access code, which Orca doesn't give me an option for. If I try to add a printer in Orca, the only option is with PIN code, which looks like this.
> 
> [image]
> 
> I'd say it's expecting a 6 character or digit code, which the X1C doesn't show me. I checked in each screen available on the printer's interface and there's no sign of a code that fits. I'm not on the beta firmware, but I am on the most recent released version at this time which is 1.08.02.00. According to Bambu's post regarding changes, they go live with 1.08.03.xx firmware which is still a beta release. Unless I'm just missing something I don't see a way to add my X1C LAN only to Orca.
> 
> I'd be happy to provide any extra info if it helps.
> 
> Edit: Since network discussions have happened, I did also put my printer and desktop on the same VLAN as a test just to eliminate any chance of connectivity issues. It still seems that the X1C only generates the access code and not the PIN code. If I have any luck with anything I'll update again.
> 
> Edit 2: I did also make sure I was signed out of my Bambu account as well, no luck there.

You need to sign out of the cloud in Orca. Logout on printer. The 6 digit code only comes up on a new connection on printer menu, and shows for only a couple minutes. Follow the link  for the "can't find this pin" it shows where it is.

alaneb, 2025-01-21:
I verified I wasn't signed in on Orca or the X1C either.  No PIN code, at this point I'm assuming unless someone knows some other trick that the X1C doesn't generate a PIN, only the access code instead, which isn't the same as the PIN.

B0IIE, 2025-01-22:
> I verified I wasn't signed in on Orca or the X1C either. No PIN code, at this point I'm assuming unless someone knows some other trick that the X1C doesn't generate a PIN, only the access code instead, which isn't the same as the PIN.

This is also true for my P1S. The printer only gives me a 8 digit acces code and Orca Slicer only accapts a 6 digit PIN.

StainlessSteele, 2025-01-22:
> This is also true for my P1S. The printer only gives me a 8 digit acces code and Orca Slicer only accapts a 6 digit PIN.

I was able to get mine working, my P1S. I had to factory reset it, and I used to setup from SD card to avoid using the Handy app, then in the account page, it shows the longer access code. I also ensured I was signed out. It works fine. https://wiki.bambulab.com/en/p1/manual/p1-sd-card-network-configuration-guide

eolder, 2025-01-22:
> The ability to bind via PIN or IP and Access code has never been implemented for the X1C. Those methods were promised for a future firmware update (>v1.08.02). You cannot use either method in Studio with an X1C. So it does not matter to the X1C user if OrcaSlicer offers those options or not.

FWIW (and YMMV) I believe it's broken on the current P1 firmware as well, see my post and logs upstream in support of this claim - https://github.com/bambulab/BambuStudio/issues/5006#issuecomment-2532562443

rocket59, 2025-01-22:
Bummer. I do second this proposal https://github.com/SoftFever/OrcaSlicer/issues/1958#issuecomment-2600820053 to incorporate code like this into OS so we are no longer dependant on bambu to implement IP binding (plus I would still prefer to stay on 1.07 on my P1S for the foreseeable future).

Noisyfox, 2025-01-23:
The ability of binding the printer via IP will be added soon.

rocket59, 2025-01-23:
Awesome!

Noisyfox, 2025-01-23:
Hi all, please give this a try. You could download the app from the Articact drop-down on this page:
https://github.com/SoftFever/OrcaSlicer/pull/8148/checks

eolder, 2025-01-23:
Wow that was quick! I tested it and unfortunately I'm seeing the same issue as upstream (not shocking, since it's the same networking plugin).

alaneb, 2025-01-23:
That was quick!  I've downloaded and tested.  Signed out of my Bambu account on my X1C and switched to LAN only mode, and the new build allowed me to add it with an access code.  For reference I'm not signed in to any account in OrcaSlicer either.  I'll run a test print after some morning meetings at work and if there's no issues then I'd say it's looking good.  
It also recognizes my attached AMS.

jrentsch, 2025-01-23:
Is there any workaround for having the printer on a different subnet? 

[image]

EchoJunction, 2025-01-23:
> Is there any workaround for having the printer on a different subnet?
> 
> [image]

Same issue here. The dialog window closes itself after having presumably connected. It did not connect however. Mine is also on a different subnet (isolated VLAN).

Noisyfox, 2025-01-23:
https://github.com/bambulab/BambuStudio/issues/4512
So even BBS has this exact same issue unfortunately.

B0IIE, 2025-01-24:
> Hi all, please give this a try. You could download the app from the Articact drop-down on this page: https://github.com/SoftFever/OrcaSlicer/pull/8148/checks

I just installed the 2.3.0 - dev, set my P1S into LAN Mode, and it was automatically recognized. Before that my printer was connected via cloud.

Benny-Git, 2025-01-24:
While reading the various issues connected to this, it looks like the basic issue being that Bambu Studio (and subsequently Orca Slicer) suffer from the issue, that even if you add the printer's IP address manually, it still requires an SSDP notification to be received from the printer. This doesn't work without extra effort if your printer is on a different subnet/VLAN, so for now I use my own SSDP faker script.

Not sure if Orca Slicer could easily remove that part, or if this needs to be fixed upstream first? Issue would be https://github.com/bambulab/BambuStudio/issues/3272

Noisyfox, 2025-01-24:
> While reading the various issues connected to this, it looks like the basic issue being that Bambu Studio (and subsequently Orca Slicer) suffer from the issue, that even if you add the printer's IP address manually, it still requires an SSDP notification to be received from the printer. This doesn't work without extra effort if your printer is on a different subnet/VLAN, so for now I use my own SSDP faker script.
> 
> Not sure if Orca Slicer could easily remove that part, or if this needs to be fixed upstream first? Issue would be [bambulab/BambuStudio#3272](https://github.com/bambulab/BambuStudio/issues/3272)

This cannot be fixed without an update to the printer firmware and the network plugin. Unless we want to bundle tools like [this](https://github.com/Rdiger-36/StudioBridge) into Orca, and even if we did, it's still a terriable user experience (because user have to manually enter the details like printer model, serialnumber etc).

Benny-Git, 2025-01-24:
I'm not too knowledgeable about the internal workings of Orca, so I didn't know this is part of the network plugin only (which I guess we load as a black box from Bambu?).

While I agree it is a terrible user experience, relying on manual faker scripts or tools is _even worse_, as they need to run separately. Given that this use case only affects a subset of users running their printer in a separate network, the UI could be hidden behind an "add manually" button, so I think it would still be desirable for those users. Adding the serial, model and IP would be a one-time thing, after which Orca could store that setting and reuse it, I think. Also, this would give Orca a big Plus compared to BS, as they don't have that functionality yet :wink:

dMopp, 2025-01-24:
LAN-Only with Orca and the mentioned DEV Build is kinda working. My problem: it takes a random amount of approaches until Orca is connected to the printer. But if --> It works like the cloud connection

alaneb, 2025-01-25:
So that build that @Noisyfox  posted that allowed me to add my X1C via LAN in Orca, it did ask for printer model, serial, etc in addition to the access code, but it has been a one time thing.  The printer remains listed as an available printer since then, so as a one time annoyance it's not a bad one.

rocket59, 2025-01-25:
@alaneb was the printer on same or different subnet to your OS PC?

alaneb, 2025-01-25:
It was on the same subnet for initial testing, and currently still is.  I plan this weekend to try moving it to an isolated subnet and creating network rules to allow for the SSDP traffic.  I did some monitoring of traffic with tcpdump and saw that traffic occur regularly.  I'll post my results once I complete that test this weekend.

cryptonym64, 2025-01-25:
Running on P1S with 1.07.00.00 and Orca 1.3 dev on Windows from @Noisyfox linked above.

Was able to connect using the IP + access code with printer on same subnet.

However, Orca forgets it is connected at each startup. I have to go to Devices tab and reselect it from the printer list. The good news is it appears to no longer lose the code and hasn't reprompted.  Woud be good if it retained the connect printer when it starts.

alaneb, 2025-01-25:
@cryptonym64 This is the same way mine behaves as well.  Not forgotten, just needs to be selected each time.  
Using the Linux AppImage build for reference.
On an Ubuntu 22.04 based distro.

cryptonym64, 2025-01-25:
> This cannot be fixed without an update to the printer firmware and the network plugin. Unless we want to bundle tools like [this](https://github.com/Rdiger-36/StudioBridge) into Orca, and even if we [...]

Here is another possible kludge around the SSDP issue.  https://gist.github.com/Alex-Schaefer/72a9e2491a42da2ef99fb87601955cc3 in case you hadn't seen it. It is python based rather than Java.

Noisyfox, 2025-01-25:
> However, Orca forgets it is connected at each startup. I have to go to Devices tab and reselect it from the printer list. The good news is it appears to no longer lose the code and hasn't reprompted. Woud be good if it retained the connect printer when it starts.

Ah interesting, this might be possible to achieve, I'll take a look soon.

alaneb, 2025-01-25:
On the subject of the printer being "forgotten" I have observed that once I launch Orca and go to devices, it takes a moment for the printer to be detected.  Is Orca listening for that SSDP traffic outside of the devices screen?  I'm assuming this takes place via the Bambu network plugin somehow, feel free to correct me if I'm wrong about where that communication takes place.  If it's not listening until you're on the devices screen, could that be part of why it's being "forgotten" until you go there?

cryptonym64, 2025-01-25:
I had said forgotten, when in truth it is still "listed" but I have to
manually click it on device tab each time Orca starts. It's not really
totally forgotten, but sure would be nice to not have this extra step. The
cloud kept the active device between Orca restarts. Hopefully just needs
cached somewhere for it to reconnect each time as the 2.3.0 dev fixed my
issue of it losing the code. Getting there. Devs are making great progress
quickly on the mess Bambu has created.

alaneb, 2025-01-25:
@cryptonym64 That's the same thing mine is doing right now, so I was really just wondering when it queries for a known printer to see if it's online.  Like you said, it may be something simple, and I agree, the devs are making amazing progress in dealing with the whole mess that Bambu created.

Jwn5k, 2025-01-25:
I am glad to see lots of engagement here on my submission, I have installed the latest Orca 2.3.0-Dev build to stay up to date as much as possible if there have been any changes. I don't have the programming skill or knowledge to know how it would work, but @Noisyfox (or anyone too i guess) do you know what the difficulties of adding the Bind Over LAN with the 8 digit code from the the X1 Carbon printer would be? what would be necessary to do it? What is the roadblocks for doing so? I am very curious to know about it, if it's even possible or not, I hope it is as on the surface it sounds like it mitigates some of the issues that I brought up in the main post.

cryptonym64, 2025-01-27:
I previously reported that 2.3.0-dev change by @Noisyfox had gotten rid of prompting for the access code on my P1S. This morning, with no changes, it prompted for access code again. The code was confirmed as being the same as it was before, and once I typed it in, it connected. So, it seems to be perhaps intermittent in nature?

Aug291997, 2025-01-29:
Thank you to everyone on their contributions toward sorting this mess out. I uninstalled 2.2 and installed 2.3.0-dev but after I enter IP and access code it says "connecting to printer" then the window closes and the printer is not listed under devices.

closed 2025-01-31 by SoftFever

Noisyfox, 2025-01-31:
Hi everyone, this PR should solve the binding problem with printer in a different subnet https://github.com/SoftFever/OrcaSlicer/pull/8256
Please give it a try and see if it works for you, thanks!

alaneb, 2025-01-31:
Thanks!  I'll give it a try this evening!

eolder, 2025-01-31:
Thanks, just tested and this release worked for me!

Noisyfox, 2025-02-01:
> Thanks, just tested and this build worked for me!
> 
> edit: if you type the serial wrong (or in lowercase) it will crash.
> 
> ```
> Exceptions:
> Exception Code :c0000005 ACCESS_VIOLATION
> Exception Flag :0x0 NumberParameters :2 
> Param 0 :0x0 
> Param 1 :0x10 
> Context :00000085FC4FF0C0 
> ContextFlag : 0x10005f, EFlags: 0x10206 
> Fault address:  0xCA3397E1 0x1:0x1687E1 C:\Users\User\AppData\Roaming\OrcaSlicer\plugins\bambu_networking.dll
> ```

I don't think I could do much about it unfortunately.

Jwn5k, 2025-02-01:
> Hi everyone, this PR should solve the binding problem with printer in a different subnet [#8256](https://github.com/SoftFever/OrcaSlicer/pull/8256) Please give it a try and see if it works for you, thanks!

Thank you very VERY much @Noisyfox and everyone else involved with making this happen.

This is exactly what  I was asking for and I am glad my idea/thought/concern/request was deemed practical enough to get implemented into Orca Slicer (even if it's in just the -dev build for now), and I hope everybody who needed such a feature is able to use it without issue, I just tried it myself and it works as intended with my X1 Carbon, amazing job!

[3 more replies]

PR #8148 (2025-01-31): Update BBS network plugin to 1.10.1
This PR ported the following changes based on BBS 1.10.1 codebase:

- The ability of binding printers via IP (Fix #8099)
- The ability of setting AMS filaments during print (Fix #7882)
- Some other related fixes and improvements

Thanks BambuLab for those improvements!

~~Please note: with this update, we will no longer be able to streaming the live camera through cloud, only through Lan (even if the printer is not in Lan mode). At least that's what I saw with this PR and also #8103, more tests & feedbacks on this are needed.~~ Update: nvm, I missed a commit [... cut]