OrcaSlicer/OrcaSlicer #8104: Calibration menu greyed out (Windows 11, Bambu p1p)

JulianKingman, 2025-01-19:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Bambu Lab P1P

### How to reproduce

1. Add Bambu P1P via account login

### Actual results

Calibration menu is greyed out
(not the bambu native one, just the additional calibrations)
[image]

### Expected results

I can access the extended calibration tests

### Project file & Debug log uploads

Project N/A

[debug_network_Sun_Jan_19_09_52_09.log.zip](https://github.com/user-attachments/files/18469604/debug_network_Sun_Jan_19_09_52_09.log.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

### Anything else?

Similar potentially related issue, although this seems to be about the other way around (disabled bambu calibration menu in non-bambu printer): https://github.com/SoftFever/OrcaSlicer/issues/2197

MxBrnr, 2025-01-19:
You have to open a project before the calibration menu is available.

JulianKingman, 2025-01-19:
Wow 🤦‍♂ Thank you!

closed 2025-01-19 by JulianKingman