OrcaSlicer/OrcaSlicer #8106: Add setting to prevent loading printer information/project settings from .3mf

suchmememanyskill, 2025-01-19:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

(Note: I have searched keywords related to 'load', 'open', 'printer profile', but couldn't find any existing open enhancements)

.3mf files downloaded from makerworld usually contain the printer profile, filament profile and speed profile they've been created with. When you have a different printer, you have to manually switch off this forced upon new printer and settings. I would prefer to be able to choose if i want to load the settings associated with a .3mf file.

I can't find a feature in the preferences to disable this behaviour. If this feature already exists but under a different menu, it may be a good idea to put this feature in a more sensible menu for it.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

I think in the preferences screen, a good addition would be an option to change this behaviour. Possibly have the label 'Loading behaviour project (.3mf) printer settings'. I think this option would be best suited to be a dropdown with 3 options. 'Always apply', 'Always ask', 'Never apply'. 

Always apply should replicate the current behaviour. If you load a .3mf file with a printer profile included, it should load this printer profile, and replace your currently selected printer profile with the loaded one. This should be the default option.

Always ask should bring up a popup box that asks the user if they want to load the printer profile. If 'Yes' is pressed, the same behavour as 'Always apply' should be applied. If 'No' is pressed, the same behavour as 'Never apply' should be applied.

Never apply should not load the printer profile from the .3mf, keeping the current printer profile.

### Describe alternatives you've considered

Currently, i use an external python script to modify the .3mf before loading it into orca. This solution isn't ideal as it requires an extra step before loading into orca. If you do accidentally load the file, your settings will be changed without asking. Also, open slicer integrations from websites (thingiverse, printables, makerworld) leave no room for an external script to run.

For those interested, this is the python script i use. I haven't double checked if slice_info.config needs to be removed.
```py
import sys, os, zipfile, time

def remove_from_zip(path : str, remove_list : list[str]):
    temp_path = os.path.join(os.path.dirname(path), str(time.time()))

    with zipfile.ZipFile(path, 'r') as zin:
        with zipfile.ZipFile(temp_path, 'w') as zout:
            zout.comment = zin.comment

            for item in zin.infolist():
                print(item.filename)
                if item.filename not in remove_list:
                    zout.writestr(item, zin.read(item.filename))

    os.remove(path)
    os.rename(temp_path, path)

if __name__ == "__main__":
    if len(sys.argv) <= 1:
        print(f"Usage: python3 {sys.argv[0]} path/to/3mf")
        exit(1)
    
    if not os.path.isfile(sys.argv[1]):
        print("File does not exist")
        exit(1)

    remove_from_zip(sys.argv[1], ["Metadata/project_settings.config", "Metadata/slice_info.config"])
```

Noisyfox, 2025-01-20:
As a workaround, you can put a cube (or anything) on to the plate, then drag & drop the 3mf file in orca, which will bring up a dialog asking whether to import model only.

robertb945, 2025-01-20:
> As a workaround, you can put a cube (or anything) on to the plate, then drag & drop the 3mf file to it, which will bring up a dialog asking whether to import model only.

This is what I typically do, but it would be nice to have controls that would allow presets for preventing loading the filaments, profiles, etc.  Or having the option to disable them when loading the 3mf.

Noisyfox, 2025-01-20:
> This is what I typically do, but it would be nice to have controls that would allow presets for preventing loading the filaments, profiles, etc. Or having the option to disable them when loading the [...]

Yeah that's what we want to do too, however currently we still haven't made a clear decision on how this should be done, so that ppl have both full control on how the app should behave when loading a 3mf file, while still easy to use.

suchmememanyskill, 2025-01-20:
> however currently we still haven't made a clear decision on how this should be done

What design needs to be done for this?

Is my outline of the feature (a setting for always load, always ask and never load) not complete, or is missing something?

(Or are you talking about if people want more specific handling? What more control could be given?)

Noisyfox, 2025-01-20:
> What design needs to be done for this?
>
> Is my outline of the feature (a setting for always load, always ask and never load) not complete, or is missing something?
>
> (Or are you talking about if [...]

Such as what's the default behavior it should be? Should it always pop up a dialog when you opening a 3mf? etc. Basically user experience stuff.

suchmememanyskill, 2025-01-20:
Ah, i mentioned that in my original post

> Always apply should replicate the current behaviour. If you load a .3mf file with a printer profile included, it should load this printer profile, and replace your currently selected printer profile with the loaded one. This should be the default option.

Imo current behaviour shouldn't be changed unless there's a really good reason to. Changing behaviour like this could break people's workflows. Opt-in allows people to choose to use this new feature. If people really really like it, we can consider changing the default.

> Should it always pop up a dialog when you opening a 3mf?

In what context do you mean this? Imo if you select the option 'Always ask' you should always be asked, unless the .3mf doesn't contain project settings.

suchmememanyskill, 2025-01-20:
> When you select a different printer, you are asked if you wish to use (transfer) the settings that differ from your printer profile. That is essentially what you are asking for: "I would prefer to be able to choose if i want to load the settings associated with a .3mf file."

That is not at all what i'm suggesting. I'm suggesting it doesn't switch/apply those settings stored in the .3mf in the first place. Applying printer settings for different printers could even be dangerous to said printer, depending on the changes.

> Those settings are what you requested by downloading the model as a 3mf. If you want to use your own settings, download the STL/CAD file.

Makerworld sadly has quite a large amount of models where the stl/cad are not included with the post. Just a .3mf file

suchmememanyskill, 2025-01-20:
> Your proposal would require choosing to immediately accept/change the printer settings which could lose important information critical to the success of the project, (wall loops, supports, infill, etc).
>
> Current method lets you change the printer and then see what settings are different from your defaults before making your own decision about their appropriateness.

Yeah, that's why i'm suggesting this change. For others, if that's the way they want it to work, that's fine. I do not. I would prefer to apply my own settings to the model, so i don't mind them being discarded. That's why i'm suggesting having the choice to apply them or not.

Does a simple model really require overriding settings, making me switch back, even though no changes to the profile were done?

> In which case you can simply discard the settings (instead of transfer) when you select your printer.

This is simply an extra interaction (move to dropdown, click, select printer, click, move to discard, click) i do not want to do every time.

> If you want to print the model without reviewing the settings just print through Handy, letting Bambu send the g-code direct to your printer. No need to involve a slicer at all.

I don't trust other people's sliced files, nor bambu's cloud. I don't have access to the Handy app, nor do i want to.

Glodigit, 2025-01-30:
I am also on the camp of ".3MF is a (multimaterial) object file" and not the current implementation of ".3MF is a slicer save-state file". The sentiment seems to be shared in #2614.

jomixlaf, 2025-03-27:
> Imo current behaviour shouldn't be changed unless there's a really good reason to. Changing behaviour like this could break people's workflows. Opt-in allows people to choose to use this new feature. If people really really like it, we can consider changing the default.
> 

This is exactly what happened to me today. I've been having issue for a couple days now were my settings are no longer save and previously working one are gone. 

Here's what happened I opened a working 3mf file with Orca 2.3 containing customs settings, plate, name and body inside a component, colors filament ect... 

I wanna import another parts in this project. In Fusion 360 I export directly to Orca in the project already loaded a 3mf component, 

[image]

Now in Orca it say's the following 

[image]

So I click import only geometry only and ☑ remember my choice because this is what I am expecting to do when I **IMPORT** parts into my project. I always want the Geometry, or it will open a new project with the previous setting from the already opened file.

So I do my things and send it to the printer and everything is fine. 

When I open any of my previous working projects, I notice all the settings are gone; I don't save, close and reopen the same thing. Panic Attack in progress to start a client project you start rebuilding some setting you barely remember and then you save. Now the working settings that were previously not loaded are being replaced with what you think are good settings...

I continue so later on I reopen my file again, and the settings are gone... again! Any file opened with this setting mistakenly use Render My Project broken 

You see where I'm going with that. This is a big problem, now I know, but I'm gonna have to explain to my client that I must wait for me to recover partially or hopefully all working settings for their project. 

Since Orca save all projects as 3mf by default can you imagine how bad this is?

I don't mean to sound rude at all; I'm mad, but I appreciate anything being developed by the developer, and if I missed something, let me know. I'll be more than grateful to learn from my mistake, 

I posted a video on YouTube. 
https://youtube.com/shorts/HrtK3VnwrXg

Thank you for your time.

suchmememanyskill, 2025-03-27:
@jomixlaf not meant to be rude, but, that is kind of exactly what you'd expect out of such a setting, no? It does what it says on the tin, it doesn't load settings from a .3mf.

I don't think you can reasonably differentiate between files you saved yourself and external/online models, and load them differently...

jomixlaf, 2025-03-27:
Hi,

Thanks for replying suchmememanyskill

While your statement makes sense, the end result doesn't. Something that was working before no longer works as expected. This is confusing because the action that triggered this is **import** 3mf, not open 3mf. There should be a way to minimize the risk of deleting working data inside a project by doing another action, such as import.  

Again, all project for anyone using Orca are saved in 3MF format, in what world would people always reload geometry when opening a project, they would lose all their data this doesn't make sense.

The idea and flexibility is 100% fine, but the implementation lacks of security to avoid lost data when **Opening** a Project

The behaviour should be If **import** a 3mf file into projects 
"Always load geometry"
"Always ask"
"Ask when relevant"

If people chose to "Remember my choice" it should affect the import side and not the load project side, and this what confused me because I was importing a file not "File> Open Project"

Maybe separate both actions with their distinct decision. 

Let's say Orca is not running, you double-click to open a 3mf file and the same menu proposed above can be ask  
"Always load geometry"
"Always ask"
"Ask when relevant" _(maybe changing this in normal or previous behavior ?)_  I don't know
And if people choose to remember their choice it would be 100% safe.

Does it make sense what I am trying to explain? 

To recapitulate, 
        open 3mf project by default without config, with config, always ask,
        Import a 3mf by default without config, with config, always ask,

I'm French Canadian and my English is my barely second language. Thanks, Grammarly

Thanks again for your time.

suchmememanyskill, 2025-03-27:
I understand now.

I don't entirely agree with the workflow breaking part; you did tell orca to load geometry only when loading .3mf.

You probably want to open a new issue for this.

Do note currently, when the plate is empty, orca will open a project. If it isn't, then it will import/add it.

This does, in my eyes, cause inconsistent behavior when importing models from say, makerworld. If your plate didn't happen to be empty, you would suddenly load settings. This is exactly why i implemented the feature like i did. I don't think there should be a difference when you load a .3mf one way or another.

One possible and easy fix for this could be to add another setting to import geometry only on import/add only. But that may be a confusing option for an end user.

Noisyfox, 2025-03-28:
Let's continue the discussion here https://github.com/SoftFever/OrcaSlicer/issues/9108