OrcaSlicer/OrcaSlicer #8114: User Themes

christmaswreath, 2025-01-20:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

I don't love the teal that OrcaSlicer is currently locked to.

### Which printers will be beneficial to this feature?

Others

### Describe the solution you'd like

You should be able to change the accent color of OrcaSlicer.

### Additional context

Similar to #3520

tlhintoq, 2025-01-21:
Have you tried flipping it in to dark mode to see if you like that better?

[image]

christmaswreath, 2025-01-21:
It's specifically the accent color (the teal) that I dislike. The Bambu Studio color scheme is much more pleasant in my eyes. The main reason I use OrcaSlicer instead of Bambu Studio is because of the viewfinder.

christopher-lambe, 2025-02-16:
I also dislike the teal accent colour. There are several teal colours used in OrcaSlicer. The main one being RGB HEX 009688.

A quick search in the code shows there are 276 files to change where this colour is "hard coded".

https://github.com/search?q=repo%3ASoftFever%2FOrcaSlicer+009688&type=code&p=1

Bambu Studio was not set up for themes. If you really hate the colour then download the source and use notepad++ to quickly search and replace (including subfolders) using your favourite accent colour. Bambu Lab accents on 00AE42. I guess SoftFever did not want to invest the time to make the colours themable when he replaced the BL green with OrcaSlicer teal.

NEOLabs-software, 2025-04-05:
i tried doing it but there are multiple colors that i am yet to find

[image]

christopher-lambe, 2025-04-05:
> i tried doing it but there are multiple colors that i am yet to find
> 
Well done mate. That looks great. I did not get around to trying it yet as I had a space problem om my laptop. Just installed a new NVME module yesterday and I will give it a go at Easter.

I love Kodi with the standard estuary skin and pink
[image]

Was it as easy as I though it would be? or did you have some hiccups along the way?  For the colours you did not find - did you use  the eyedropper tool to identify the offending colour HEX code? I mentioned " There are several teal colours used in OrcaSlicer"

NEOLabs-software, 2025-04-05:
i did use xcolor, which is just an eyedropper tool made for x11
although some colors i had to manually find in src/slic3r/GUI/BitmapCache.cpp as mentioned in https://github.com/SoftFever/OrcaSlicer/issues/3520

i made a script and dropped it in src/slic3r/GUI that goes something like
```
#!/bin/bash

DIR="."

OLD="52c7b8"
NEW="a62699"

find "$DIR" -type f \( -name "*.svg" -o -name "*.cpp" -o -name "*.xml" -o -name "*.css" \) | while read -r file; do

    sed -i "s/$OLD/$NEW/g" "$file"
    if [ $? == 0 ]; then
    echo "Processed: $file"
    fi
done

echo "Replacement complete!"
```

i ran it for each color and replacement i found, compiled it and tested
and for the rest, they are just images or css files found at the resources directory

good luck retheming

NEOLabs-software, 2025-04-05:
oh, and some colors are in this weird rgba format, so for the main color, replace ```0.0f, 150.f / 255.0f, 136.0f / 255```

christopher-lambe, 2025-04-18:
> i made a script and dropped it in src/slic3r/GUI that goes something like
>  SNIP
> find "$DIR" -type f \( -name "*.svg" -o -name "*.cpp" -o -name "*.xml" -o -name "*.css" \) | while read -r file; do
> SNIP 
> i ran it for each color and replacement i found, compiled it and tested and for the rest, they are just images or css files found at the resources directory
> 
> good luck retheming

I finally got OrcaSlicer to compile on Windows. A few questions if you have time:
There are no svg or xml or css files in the directory and subdirectories of src/slic3r/GUI. Where did you run the script?
Can you post your "//map of colour replaces" section from OrcaSlicer\src\slic3r\GUI\BitmapCache.cpp?
Can you also post the path to one file that has the "weird rgba format" you mentioned?

NEOLabs-software, 2025-04-18:
The svg and csa is found in the resources directory at the root of
orcaslicer

And the float rgb is found in multiple files
I used the script and ChatGPT to translate the float to hex and imputed it
into the script

I will send the bitmapcache soon as well as the format for the float

(Have in mind, there is still some disgusting teal left, as I have not
found the colors corresponding to it)

PandaBalls, 2025-04-22:
[image]
I am a color-blind user, and I simply cannot tell whether the LED light is on or off.  
我是色弱用户，我根本无法看清LED灯是开着还是关着的。

christopher-lambe, 2025-04-23:
> The svg and csa is found in the resources directory at the root of
> orcaslicer
>
> And the float rgb is found in multiple files
> I used the script and ChatGPT to translate the float to hex and imputed it [...]

Thanks NEO. Looking forward to seeing the files from you. It will be interesting to see how many different teals shades there are in the end when we find them all. I will document them and post them here when I am finished.

christopher-lambe, 2025-04-23:
> [image] I am a color-blind user, and I simply cannot tell whether the LED light is on or off. 我是色弱用户，我根本无法看清LED灯是开着还是关着的。

Which colour blindness do you have?  https://paireyewear.com/blogs/news/types-of-color-blindness.
I am guessing the second one in the list above.
If it helps you you can change the svg images with notepad++ as I mentioned in a post above.

C:\Program Files\OrcaSlicer\resources\images\monitor_lamp_on.svg
You can change the colours in the svg files without compiling OrcaSlicer. Maybe it helps you.