OrcaSlicer/OrcaSlicer #8115: Unable to set custom camera stream

mLupine, 2025-01-20:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0 nightly (5ec8f51)

### Operating System (OS)

macOS

### OS Version

15.3 (24D5034f)

### Printer

A1 / X1C

### How to reproduce

1. Open up the device tab and select any Bambu printer
2. Click on the "camera settings" button on top of the camera stream (first one on the right)
3. Try typing anything into the "Hostname or IP" field

### Actual results

The text field is inactive (and so is the button next to it), there's no way to type anything into it

### Expected results

I should be able to type in anything

### Project file & Debug log uploads

Happens also when there's no project selected. No related logs either, it's a UI issue.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

ksignorini, 2025-01-24:
Same problem. Just wanted to show it's not isolated to one user.

larcin78, 2025-02-02:
I have the same issue as well. Click in the box and attempt to enter an IP and the dropdown closes as soon as you try to type or copy anything.

mLupine, 2025-02-02:
There’s a rather simple (although still annoying) workaround I found. 

1. Open the camera settings popup. If the buttons and fields are active, you’re in luck. If not:
2. Without closing the popup, select another tab from the sidebar on the left (like HMS)
3. Go back to the first tab and open the camera settings popup. It should now be working fine for the next few minutes. 

As a side note, it’d be really convenient to be able set a custom camera stream just for one printer and have it go back to the Bambu stream (or another custom one) after selecting another printer in the Device tab.

ksignorini, 2025-02-03:
> There’s a rather simple (although still annoying) workaround I found.
>
> 1. Open the camera settings popup. If the buttons and fields are active, you’re in luck. If not:
> 2. Without closing the popup, [...]

Nah, doesn't help. I did get the cursor in the text field, but as soon as I tried to type, it disappeared again.

rikahlberg, 2025-02-11:
> There’s a rather simple (although still annoying) workaround I found.
>
> 1. Open the camera settings popup. If the buttons and fields are active, you’re in luck. If not:
> 2. Without closing the popup, [...]

> Nah, doesn't help. I did get the cursor in the text field, but as soon as I tried to type, it disappeared again.

Doesn't work for me either with the 2.2.0 release on MacOS. I get a cursor in the "Hostname or IP" field, but the dialog disappears as soon as I press a key.

jessedmatlock, 2025-03-05:
Same issue, running 1.10.1.150 on Mac OSX 15.3.1
Cannot type into the Camera URL dropdown, it closes. I CAN right click and paste.. but even then the Camera feed shows as a white/blank box.. no love.

ebarinia, 2025-03-20:
Same issue on my side. Unable to add an external feed or use external camera on latest MacOS 15.3.2 and Orcaslicer 2.3.0-rc build 79a5a9b.

closed 2025-06-27 by github-actions[bot]