OrcaSlicer/OrcaSlicer #8122: Feature request: Hide other plates... or Hide other objects

tlhintoq, 2025-01-21:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

Feature request: `Hide other plates`

When working on one plate of many, especially when it is a plate surrounded on all side by other plates, it can be really hard to see and maneuver.

For example: Often times trying to place text on one object is near impossible when the view is obstructed by other objects.

[image]

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

On the current plate, right side tools for that plate, add a "hide others" button below the "move to front" button.

### Describe alternatives you've considered

Alternatively... All the tools in the `Prepare` tab top tool bar should act the same.
When you select the `Z seam` tool, all other objects are hidden.  Same when you select the `Supports` tool.  Maybe just continue that behavior to all tools of that tool bar including the text tool.

### Additional context

discip, 2025-01-22:
This could possibly be solved by switching the **`Emboss`** **gizmo** to the mode used by **`Support Painting`**, **`Seam Painting`** and **`Color Painting`**, which hides everything except the selected part.

tlhintoq, 2025-01-23:
> This could possibly be solved by switching the **`Emboss`** **gizmo** to the mode used by **`Support Painting`**, **`Seam Painting`** and **`Color Painting`**, which hides everything except the selected part.

Yes.  Exactly.  That's what I meant when I described that all of the tools in that tool bar should behave the same.

discip, 2025-11-05:
> [@tlhintoq](https://github.com/tlhintoq) it looks like this is fixed, could you close?

That is unfortunately not the case:

[image]

coryrc, 2025-11-05:
> > [@tlhintoq](https://github.com/tlhintoq) it looks like this is fixed, could you close?
> 
> That is unfortunately not the case:

My fault for misreading the mention. I thought it was a PR OP made.

discip, 2026-09-24:
@ianalexis
@RF47
Is it hard to implement this?

Please let me know if you'd prefer not to be tagged. 😊

RF47, 2026-10-10:
> 9\. [@ianalexis](https://github.com/ianalexis)
>     [@RF47](https://github.com/RF47)
>     Is it hard to implement this?
>     Please let me know if you'd prefer not to be tagged. 😊

my attempt #16325

closed 2026-10-14 by ianalexis