OrcaSlicer/OrcaSlicer #8124: Allow Naming Conventions for .3mf files

Warweedy, 2025-01-21:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

When sending print jobs to the printer .3mf files get created and sent. The plate number gets added with underscores.
There is no other information included in the file name.
When creating gcode, there is the option to define a naming scheme. This should also be available for the .3mf files.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

It should be possible to define a Filename with variables for .3mf files or files sent with "Print plate".

tlhintoq, 2025-01-22:
You marked this as 'all printers' - but its really specific to printers that take a .3mf file as their print file, right?
That's not `all` printers.  Most printers require a .gcode not a .3mf as their printable file type.
So what printer specifically are you using that takes a .3mf as the printable?

Warweedy, 2025-01-22:
Sorry for my confusion i thought the [image] option always sends .3mf files. 
I am working with a BambuLab A1 Mini, so i guess it at least is BambuLab specific?