OrcaSlicer/OrcaSlicer #8134: AppImage (non-ubuntu) not working, missing libwebkit library

ih8ms, 2025-01-22:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.2.0, 2.3.0-dev, both AppImage in Non-Ubuntu flavor

### Operating System (OS)

Linux

### OS Version

Arch

### Printer

Bambu Lab A1 mini

### How to reproduce

- get AppImage in Non-Ubuntu flavor
- try to start Orca via AppImage

### Actual results

Error message:

`/tmp/.mount_OrcaSlcn2MTB/bin/orca-slicer: error while loading shared libraries: libwebkit2gtk-4.0.so.37: cannot open shared object file: No such file or directory`

### Expected results

No error message but Orca slicer starting

### Project file & Debug log uploads

- no project file, no project involved
- no log file is written

### Checklist of files to include

- [x] Log file
- [x] Project file

### Anything else?

https://github.com/SoftFever/OrcaSlicer/issues/4616, marked as closed, but obviously not fixed?

liberodark, 2025-01-22:
Hi,

A workaround is to use flatpak directly.

Best Regards

DarthRidar, 2025-01-24:
sudo pacman -S webkit2gtk

ih8ms, 2025-01-29:
> sudo pacman -S webkit2gtk

Isn't an AppImage supposed to be self-contained and come with all libraries needed to run?

ivakorin, 2025-02-18:
Same problem with Debian sid

brunetton, 2025-02-25:
Same here (Ubuntu 24.04)

    /tmp/.mount_OrcaSlZB2bB8/bin/orca-slicer: error while loading shared libraries: libwebkit2gtk-4.0.so.37: cannot open shared object file: No such file or directory

juli3nk, 2025-03-04:
Same problem with NixOS.